niceboygithub/AqaraM1SM2fw #1: How to use this firmware？

miniknife88, 2021-01-20:
Thanks for making the firmware

So，How to use this firmware？

niceboygithub, 2021-01-29:
Use the aqaragateway.exe to flash via UART TTL

thejonesyboy, 2021-05-07:
Aqara Hub M1S CN ZHWG15LM

I'm getting an error:
`'NullWriter' object has no attribute 'fileno'`

The hub is plugged in to mains, so I assume I don't need to connect 3.3V from TTL to the hub?

[image]

[image]

niceboygithub, 2021-05-07:
1. I update the tool.
2. 3.3V is not necessary to connect to hub.
3. use original power supply board.
4. use fw_update to flash firmware is better method.
5. switch to Mi Home mode and enable telnet. use putty to login and enter command in README.md
6. Aqara Home mode need to use modified firmware.

thejonesyboy, 2021-05-07:
Thanks for the quick reply.

1. when and why?
2. ok
3. I have it plugged into the main 240VAC power, is that what you mean?
4. I would prefer to use UART (see next item)
5.  'custom open telnet command' doesn't work, see my comment here: https://gist.github.com/zvldz/1bd6b21539f84339c218f9427e022709#gistcomment-3733592
6. Do you mean switch to Aqara home mode to boot the modified firmware after it has been flashed? I wish I could get that far :sob:

niceboygithub, 2021-05-07:
> Thanks for the quick reply.
> 
> 1. when and why?

https://github.com/niceboygithub/AqaraM1SM2fw/commit/f56610355f83beef03b8f5eabe0fca5c03b6856f
Because the progress bar is not display correctly.

> 3. I have it plugged into the main 240VAC power, is that what you mean?

Yes

> 4. I would prefer to use UART (see next item)
> 5. 'custom open telnet command' doesn't work, see my comment here: https://gist.github.com/zvldz/1bd6b21539f84339c218f9427e022709#gistcomment-3733592

I tested the way 2 and way 3. It shall be work. You need to make sure that the hub is working as Mi Home mode.
If it is still not working, try another network.

> 6. Do you mean switch to Aqara home mode to boot the modified firmware after it has been flashed? I wish I could get that far 😭

After flash the modified firmware, then switch to Aqara home mode. Otherwise the telnetd will be not enabled.

thejonesyboy, 2021-05-07:
1. OK, so it's unrelated.
5. As explained in the link I posted, I definitely have it in Mi Home mode, and the device is discoverable on the 2.4ghz-only  network via both php and python methods. It is the telnet open command that I can't figure out. Which way, 2 or 3 worked for you? Would you kindly check my post to see if there is something I'm doing wrong? https://gist.github.com/zvldz/1bd6b21539f84339c218f9427e022709#gistcomment-3733592

niceboygithub, 2021-05-14:
[image]

I use way 3 to enable telnet.
You can see the picture. After sent the command, M1S reply 'ok'. And saved the log to /data/miiocli.log after enable telnet.

closed 2021-06-25 by niceboygithub