niceboygithub/AqaraM1SM2fw #3: aqaragateway tool not available

miniknife88, 2021-01-30:
I have connected the UART TTL to the M2 gateway and connected to the PC. Use the aqaragateway tool to click on the flash buttion, nothing happens...
Just show：

{'ddr_base': '0xa1000000', 'xmodem': True, 'tftp': False, 'telnet': False, 'comport': 'COM4', 'baudrate': 38400, 'fwtype': 'rootfs_1', 'fwfile': 'E:/aqara/rootfs_3.0.2_0014_0508_modified.bin', 'model': 'M2', 'debug': False}

What should be the correct step?

niceboygithub, 2021-02-01:
If you wait a second, it will show the following messages.
```
Please power up gateway!
If your gateway is powered up, disconnect usb cable and reconnect it.
```

miniknife88, 2021-02-01:
thanks! 
UART TTL should you use 3.3V mode?
Whether to flash the modified firmware into rootfs_1？

niceboygithub, 2021-02-01:
3.3V
You can flash rootfs_0 or rootfs_1 or both.

miniknife88, 2021-02-02:
I have followed your steps, but nothing is displayed：

Please power up gateway!
If your gateway is powered up, disconnect usb cable and reconnect it.

niceboygithub, 2021-02-02:
it means that it boots up normally, did not enter download mode.
You can use putty to see if any logs from UART. Make sure that UART TX RX work probably.

miniknife88, 2021-02-02:
Use putty to get the log（Did not see the command symbol）：

`fw_manager.sh revision: 2
pre-revision: 2

hostname: Aqara-Hub-M2-5154
type: lumi.gateway.iragl5, model: AH_M2_BLE
Jan  1 00:00:09 mDNSResponder: mDNSResponder (Engineering Build) (Oct 26 2020 21:07:35) starting
Jan  1 00:00:09 mDNSResponder: Unable to parse DNS server list. Unicast DNS-SD unavailable
Jan  1 00:00:09 [    9.950000] store_tty0_enable buf=disable
[    9.950000] , count=8
mDNSResponder: mDNSPlatformSourceAddrForDest: connect 1.1.1.1 failed errno 128 (Network is unreachable)
Jan  1 00:00:09 mDNSResponder: WARNING: mdnsd continuing as root because user "nobody" does not exist
[   11.580000] genirq: Flags mismatch irq 79. 00000083 (gpiolib) vs. 00000082 (wps btn)
[   18.190000] i2c_designware i2c_designware.2: i2c_dw_handle_tx_abort: slave address not acknowledged (7bit mode)
[   24.720000] i2c_designware i2c_designware.2: i2c_dw_handle_tx_abort: slave address not acknowledged (7bit mode)
`

miniknife88, 2021-02-02:
Thank you for your patience，use gateway3utils.exe，Finally succeeded！

miniknife88, 2021-02-04:
aqaragateway.exe is still invalid for m1s gateway，cannot enter download mode，fix it please ！

miniknife88, 2021-02-04:
Thanks for the update, but the new version still reports an error:

{'ddr_base': '0xa1000000', 'xmodem': True, 'tftp': False, 'telnet': False, 'comport': 'COM5', 'baudrate': 38400, 'fwtype': 'linux_0', 'fwfile': 'E:/iot_tool/linux_3.0.4_0005.bin', 'model': 'M1S', 'debug': False}
'NullWriter' object has no attribute 'fileno'

niceboygithub, 2021-02-16:
If you are using it on M1S, need to take care the power supply of M1S.

I tested this tool on M1S and M2 without any problem.

miniknife88, 2021-02-16:
Are there requirements for the operating system, I use windows10 professional edition

I tested it on three different computers and all showed the same error

closed 2021-06-07 by niceboygithub