niceboygithub/AqaraM1SM2fw #5: Excuse me? How to update the modified firmware？

xztxy, 2021-02-24:
I downloaded your firmware, but I don't know how to operate it. Can you give me a detailed procedure?

niceboygithub, 2021-02-25:
which gateway do you use?
If M1S, you can enable telnetd. Then login by telnet, use fw_date to flash modified firmware.
If M2, you need to wire out the UART. Then use aqaraway.exe to flash it.

sergeiKokorin, 2021-06-04:
> какой шлюз вы используете?
> Если M1S, вы можете включить telnetd. Затем войдите в систему по telnet, используйте fw_date для прошивки модифицированной прошивки.
> Если M2, вам нужно подключить UART. Затем используйте aqaraway.exe, чтобы его прошить.

> which gateway do you use?
> If M1S, you can enable telnetd. Then login by telnet, use fw_date to flash modified firmware.
> If M2, you need to wire out the UART. Then use aqaraway.exe to flash it.

I also cannot flash the M1S. I tried to connect via UART and nothing came of it through aqaraway.exe. And how to enable telnetd and how to run fw_date I cannot understand. Maybe there are detailed instructions? my hub has been updated to firmware 3.2.6_0014.0517 and now it cannot update via the standard AQARA 3.2.6_0014.0524 application. another question is it possible to change the region from EU to CN

niceboygithub, 2021-06-05:
> I also cannot flash the M1S. I tried to connect via UART and nothing came of it through aqaraway.exe. And how to enable telnetd and how to run fw_date I cannot understand. Maybe there are detailed instructions? 

I remember that you enable telnet of gateway first, then use fw_update to flash custom firmware
To enable telnet, you can use the way 2 or way 3 in the [website](https://gist.github.com/zvldz/1bd6b21539f84339c218f9427e022709)
Or you can use this [integration](https://github.com/niceboygithub/AqaraGateway). Just put M1S in Mija home and find out the token of gateway.

If you switch Aqara home, you need to flash custom firmware.
You need to download [Putty](https://www.chiark.greenend.org.uk/~sgtatham/putty/).
Open putty, choose Telnet as connection type, then enter the ip of M1S.
Next step is to login with the username 'admin'.

If you get '#' prompt, enter the following command.

`
cd /tmp
wget -O /tmp/curl "http://master.dl.sourceforge.net/project/mgl03/bin/curl?viasf=1"
chmod a+x /tmp/curl
/tmp/curl -s -k -L -o /tmp/linux.bin https://github.com/niceboygithub/AqaraM1SM2fw/blob/main/modified/M1S/3.2.4_0014.0520_mi_fw_ver_3.1.3_0011/rootfs_3.2.4_0014.0520_mi_fw_ver_3.1.3_0011_modification.bin
fw_update /tmp/linux.bin
/tmp/curl -s -k -L -o /tmp/rootfs.bin https://github.com/niceboygithub/AqaraM1SM2fw/blob/main/modified/M1S/3.2.4_0014.0520_mi_fw_ver_3.1.3_0011/rootfs_3.2.4_0014.0520_mi_fw_ver_3.1.3_0011_modification.bin
fw_update /tmp/rootfs.bin
`

Then enter 'reboot' to reset gateway. After reset, you can switch M1S to Aqara home mode.

> my hub has been updated to firmware 3.2.6_0014.0517 and now it cannot update via the standard AQARA 3.2.6_0014.0524 application. 

M1S firmware 3.2.6 was removed by Xiaomi, so I did not get it before removed.
Use above the instructions, you can downgrade the firmware.

> another question is it possible to change the region from EU to CN
What you meaning? Do you mean the region of Mija Home app?

closed 2021-06-26 by xztxy