niceboygithub/AqaraM1SM2fw #9: Upgrading the firmware via fw_update is invalid

miniknife88, 2021-04-22:
M1s: 3.2.4_0014.0520
M2: 3.2.4_0010.0520

The firmware version has not changed after the upgrade is complete

niceboygithub, 2021-06-05:
which commands you use?

miniknife88, 2021-06-24:
Upload firmware to gateway tmp directory，use：
fw_update ./linux_3.2.7_0018_0524.bin
fw_update ./rootfs_3.2.7_0018_0524_modification.bin
reboot

Display firmware version: 3.2.7_0018.0515
[image]

niceboygithub, 2021-06-24:
I update the readme for flash M2 firmware for your reference.

https://github.com/niceboygithub/AqaraGateway#flash-m2-custom-firmware-method

miniknife88, 2021-06-24:
Follow the readme to flash the firmware, can't upgrade to 3.2.7_0018.0524
`# fw_update ./linux.bin
Info: Open ./linux.bin, size: 2233364
Info: Firmware type: Linux1
Info: data_sum: 8040
Info: Upgrade linux bank 0
Info: Write to /dev/mtdblock4, size:2233348 ...
Process: 100%
Done
Info: Verify ...
Process: 100%
Success`

`# fw_update ./rootfs.bin
Info: Open ./rootfs.bin, size: 9435156
Info: Firmware type: Root1
Info: data_sum: 0a89
Info: Upgrade rootfs bank 0
Info: Write to /dev/mtdblock5, size:9435140 ...
Process: 100%
Done
Info: Verify ...
Process: 100%
Success
`

niceboygithub, 2021-06-25:
fw_update said the flash is success.
What is your mean that can’t upgrade?

miniknife88, 2021-06-25:
The firmware version in aqarahome has not changed

niceboygithub, 2021-06-25:
Did you upgrade Aqara Home to latest?
Maybe it is bug of Aqara cloud service.

miniknife88, 2021-06-25:
aqarahome version is the latest：2.3.11

xfzha0, 2021-07-11:
Me too.
aqara home app version: V2.3.12 
M2 firmware 3.2.8_0006.0508 to 3.2.8_0006.0526.  
fw_update said the flash is success. 
After restart ,The firmware version in aqarahome has not changed.

niceboygithub, 2021-07-12:
You need to upgrade zigbee firmware.

update bt firmware.
```
run_ble_dfu.sh /dev/ttyS1 full.gbl 125 1
```

update zigbee firmware
```
zigbee_msnger zgb_ota ControlBridge.bin
```

miniknife88, 2021-07-12:
update zigbee firmware error:

`zigbee ota(ver: v1.0.0) firing up, built at Mar 17 2021 10:29:17; Author: liang.                                                
wang@aqara.com
connect localhost:1883
{"cmd":"write","id":2408,"did":"lumi.0","source":"","params":[{"res_name":"8.0.2                                               
 089","value":{"file_name":"./ControlBridge.bin","version":"V3.78"}}]}

on_connect result=0
ota zigbee coordinator failed, errno =8
`

niceboygithub, 2021-07-13:
> update zigbee firmware error:
> 
> `zigbee ota(ver: v1.0.0) firing up, built at Mar 17 2021 10:29:17; Author: liang.
> [wang@aqara.com](mailto:wang@aqara.com)
> connect localhost:1883
> {"cmd":"write","id":2408,"did":"lumi.0","source":"","params":[{"res_name":"8.0.2
> 089","value":{"file_name":"./ControlBridge.bin","version":"V3.78"}}]}
> 
> on_connect result=0
> ota zigbee coordinator failed, errno =8
> `

Use absolute path

miniknife88, 2021-07-21:
update zigbee firmware success，but firmware version in aqarahome has not changed.

niceboygithub, 2021-07-27:
> update zigbee firmware success，but firmware version in aqarahome has not changed.

What the version  in gateway and in aqara home?

miniknife88, 2021-07-27:
All use the latest version of firmware，after upgrading:
M2: 3.2.8_0006.0515
M1S: 3.2.8_0009.0519

closed 2021-09-19 by niceboygithub