niceboygithub/AqaraM1SM2fw #12: Can't flash M2

kolaa, 2021-08-01:
aqaragateway.exe is not working with Aqara M2

M2 current firmware version 3.2.7_0018_0524

When i trying to flash rootfs, i see

Goto flasher failed, try again.
Error: 'NullWriter' object has no attribute 'fileno'

niceboygithub, 2021-08-02:
If you are using M2 EU, the flash is not support M2 EU.

gnep88, 2021-09-18:
I had exactly the same issue with M2 CN, the EXE does not work for me as well with above issue.

In the end I flashed by using "gateway3utils.exe", now works fine.

See the CMD history for reference:

D:\aqara>gateway3utils.exe -x -c COM5 -t linux_0 -f .\linux_3.3.0_0017.0526.bin
Please power up gateway3!
If your gateway3 is powered up, disconnect usb cable and reconnect it.
Downloading the flasher.
0% [##############################] 100% | ETA: 23:59:59
Total time elapsed: 00:00:46
Now transmitting .\linux_3.3.0_0017.0526.bin_raw
0% [############################################################] 100% | ETA: 00:00:00
Total time elapsed: 00:01:46
Transmit Done! Please wait for programming to flash.
Programming .\linux_3.3.0_0017.0526.bin_raw Done!

D:\aqara>gateway3utils.exe -x -c COM5 -t rootfs_0 -f .\rootfs_3.3.0_0017.0526_modified.bin
Please power up gateway3!
If your gateway3 is powered up, disconnect usb cable and reconnect it.
Downloading the flasher.
0% [##############################] 100% | ETA: 23:59:59
Total time elapsed: 00:00:49
Now transmitting .\rootfs_3.3.0_0017.0526_modified.bin_raw
0% [############################################################] 100% | ETA: 00:00:00
Total time elapsed: 00:07:13
Transmit Done! Please wait for programming to flash.
Programming .\rootfs_3.3.0_0017.0526_modified.bin_raw Done!

niceboygithub, 2021-09-18:
> I had exactly the same issue with M2 CN, the aqaragateway.exe does not work for me as well with above mentioned issue.
> 
> In the end I flashed by using "gateway3utils.exe", now works fine.
> 
> See the CMD history for reference:
> 

NOT recommend to use gateway3utils to flash M2.

gnep88, 2021-09-19:
@niceboygithub THanks for the information. But the aqarawaygateway.exe does not work for me with error:
Goto flasher failed, try again.
Error: 'NullWriter' object has no attribute 'fileno'

Seems like a python error. Would it be possible to output more debug info?

niceboygithub, 2021-09-19:
Please plug-in USB2Uart first and have try.

zsqduke, 2021-10-03:
Hi, I have the same error code. I can see the boot sequence from putty. Do we need to do something in Putty? Like, interrupt the boot sequence?

niceboygithub, 2021-10-04:
do not open COM port with other app, like putty.

gnep88, 2021-10-05:
@niceboygithub I tried again with 3.3.2_0008.0605. First plugged in the USB-Serial then start the exe, same error on Windows 10:
Error: 'NullWriter' object has no attribute 'fileno'.

As a fallback and only solution, I have to use "gateway3utils.exe". 

If you say this is not recommended, would it be possible to provide a command line tool like gateway3utils.exe, for M2 ? Thanks

niceboygithub, 2021-10-05:
You can test UART connection by putty first.
Make sure that the logs is output from gateway. Then power off the gateway and close the putty. 
Try flash again.

zsqduke, 2021-10-05:
I did use Putty to see log so I can confirm my UART connection is working. Same error from the .exe tool no matter what I do. I have talked to Miniknife88 who had the same issue. I’m wondering what am I doing different than you

niceboygithub, 2021-10-09:
Please try the latest one