niceboygithub/AqaraM1SM2fw #13: M2 Modified FW 3.3.0_0017.0526 flashed successfully but identified as 3.3.0_0017.0515 in Aqara app

gnep88, 2021-09-18:
In the Aqara app, it still notifies a new version available: 3.3.0_0017.0526, while current version shows as 3.3.0_0017.0515.

My original FW was 3.0.7_0008_0515, maybe the last 4 digits "0515" got kept after updated ?

Command line history:

D:\aqara>gateway3utils.exe -x -c COM5 -t linux_0 -f .\linux_3.3.0_0017.0526.bin
Please power up gateway3!
If your gateway3 is powered up, disconnect usb cable and reconnect it.
Downloading the flasher.
0% [##############################] 100% | ETA: 23:59:59
Total time elapsed: 00:00:46
Now transmitting .\linux_3.3.0_0017.0526.bin_raw
0% [############################################################] 100% | ETA: 00:00:00
Total time elapsed: 00:01:46
Transmit Done! Please wait for programming to flash.
Programming .\linux_3.3.0_0017.0526.bin_raw Done!

D:\aqara>gateway3utils.exe -x -c COM5 -t rootfs_0 -f .\rootfs_3.3.0_0017.0526_modified.bin
Please power up gateway3!
If your gateway3 is powered up, disconnect usb cable and reconnect it.
Downloading the flasher.
0% [##############################] 100% | ETA: 23:59:59
Total time elapsed: 00:00:49
Now transmitting .\rootfs_3.3.0_0017.0526_modified.bin_raw
0% [############################################################] 100% | ETA: 00:00:00
Total time elapsed: 00:07:13
Transmit Done! Please wait for programming to flash.
Programming .\rootfs_3.3.0_0017.0526_modified.bin_raw Done!

niceboygithub, 2021-09-18:
you need to update Zigebee firmware.
`
cd /tmp && wget -O /tmp/curl "http://master.dl.sourceforge.net/project/mgl03/bin/curl?viasf=1" && chmod a+x /tmp/curl
/tmp/curl -s -k -L -o /tmp/m2_update.sh https://raw.githubusercontent.com/niceboygithub/AqaraM1SM2fw/main/modified/M2/m2_update.sh
chmod a+x /tmp/m2_update.sh && /tmp/m2_update.sh
`

gnep88, 2021-09-19:
Issue solved. Now I know 0526 means zigbee coordinator firmware. 
THank you very much niceboy.

closed 2021-09-19 by gnep88