niceboygithub/AqaraM1SM2fw #14: aqaragateway.exe doesn’t work

zsqduke, 2021-10-04:
I have a bricked M1S and want to flash stock firmware but can’t get your tool working. Here’s the screen recording. https://imgur.com/a/Fhjqpq0

I’m using Win10. What’s your environment in which it worked?

niceboygithub, 2021-10-04:
Please check that your COM3 was not used by any other apps.

zsqduke, 2021-10-04:
No it was not.

This is what I did. I 1) plugged in UART-USB, 2) powered on M1S and 3) opened aqaragateway.exe.

Is this how I’m supposed to do this, or is there some other steps that is obvious to you pros that I missed? Like, do I need to interrupt the boot sequence? Please be so kind as to describe it step by step.

niceboygithub, 2021-10-04:
1. Plug-in USB2UART 
2. Run AqaraGateway.exe
3. Choose firmware, flash type, etc
4. Click the Flash button
5. Wait the prompt to power up M1S

niceboygithub, 2021-10-09:
Please try the latest one