niceboygithub/AqaraM1SM2fw #16: Can't boot M2

shibechko, 2021-10-12:
Hello. please help me.
I flash my Aqara hub M2 (lumi.gateway.iragl15) with modifyed rootfs to rootfs_1.
After that gateway boot normaly, but alway close telnet session.
Then I flash modifyed rootfs to rootfs_0 and ....

`=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2021.10.13 00:02:38 =~=~=~=~=~=~=~=~=~=~=~=
uart ok

strap:0x412b8ae2

enable spi-nand

ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 MHz)

Found recognized ID, rdid=0x00efaa21

init IP fail(0xffffffff)

init ddr ok

img sig ok

chksum ok

load img ok

s-boot

sec sig ok

decrypt img

jump 0xa0000000

SPI Nand ID=00efaa21

SPI Nand die chipsize=0x08000000 byte

SPI Nand dienum=1,

SPI Nand blocksize=0x00020000 byte,

SPI Nand pagesize=0x00000800 byte,

SPI Nand oobsize=0x00000040 byte,

[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024

[rtkn_scan_bbt, line 1822] block_v2r_num 00000400

[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,

[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd

INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb

load bbt v2r table:0 page:512

[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!

check v2r bbt table:0 OK

[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704

INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb

load bbt table:0 page:704

[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!

check bbt table:0 OK

[dump_BBT] Nand BBT Content

Congratulation!! No BBs in this Nand.

Realtek Crypto Engine v0.1

=>CPU Wake-up interrupt happen! GISR=09000084 

 

---Realtek RTL8197F boot code at 2020.07.14-20:40+0800 v3.4T-pre2.2 (993MHz)

Info: Load boot_info success!

== RTL8197 Aqara Gateway bootloader ==

boot_info: ver:0

kernel: newest:0, curr:0

rootfs: newest:0, curr:0

kernel[0]: sum:0x80bc, size:2233412, fail:0

      [1]: sum:0x808a, size:2233412, fail:0

rootfs[0]: sum:0x0000, size:9445380, fail:3

      [1]: sum:0x0000, size:9445380, fail:3

root_sum_check: off

watchdog_time: 0

boot_version: 1.0.0_0001

boot_magic: 0000917c

priv mode

Info: loading kernel 0 ... size 2233412

Info: checking kernel 0 ... 

Success!

Info: rootfs 0 is invalid

Info: rootfs 1 is invalid

Warn: no rootfs available.

---Ethernet init Okay!

<RealTek>`

Thanks.

shibechko, 2021-10-16:
`
Generating padded firmware, please wait...!
The raw firmware is invaild format.
Generate padded firmware Failed!
`

niceboygithub, 2021-10-17:
> Original EU rootfs `Generating padded firmware, please wait...! The raw firmware is invaild format. Generate padded firmware Failed!`

The format of the signed firmware is different from non-signed one.
The utility of AqaraGateway.exe need to be modified to support it.

niceboygithub, 2021-10-17:
Please use the latest one to flash again.

shibechko, 2021-10-17:
Unfortunately it's does not help.
Maybe you have original EU v3.3.0_0017.0526? It was before all mods.

niceboygithub, 2021-10-18:
> Unfortunately it's does not help. 

What is the logs?

> Maybe you have original EU v3.3.0_0017.0526? It was before all mods.

I already uploaded the original firmwares.
Please do NOT flash modified firmware in EU version as I said in several days ago.

rezmus, 2021-10-18:
rootfs checksum should be 0xfbb5

niceboygithub, 2021-10-19:
> rootfs checksum should be 0xfbb5

@rezmus 
The "root_sum_check" is off. Is the sum check still working?

I did not use the correct way to calculate the sum in bootloader.
1. I lost the method you gave me to calculate the sum.  Can you give me again?
2. In CN version, the bootloader did not use the sum check.

niceboygithub, 2021-10-20:
> Unfortunately it's does not help. Maybe you have original EU v3.3.0_0017.0526? It was before all mods.

I try to figure out the solution and not 100% sure that it works.
Because without dump the rootfs which flashed by AqaraGateway.exe, it is hard to debug. 

Please download the latest AqaraGateway.exe to flash again.

shibechko, 2021-10-20:
```
Do NOT power up your gateway.
IF it is already power up, remove power cable!

Generating padded firmware, please wait...!
Please power up gateway!
If your gateway is powered up, disconnect usb cable and reconnect it.
Downloading the flasher.
Now transmitting C:\Users\shibeko\AppData\Local\Temp\root_3_3_0_0021_0526.bin_raw
Transmit Done! Please wait for programming to flash.
Programming C:\Users\shibeko\AppData\Local\Temp\root_3_3_0_0021_0526.bin_raw Done!
```
```

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2021.10.20 22:47:34 =~=~=~=~=~=~=~=~=~=~=~=
uart ok

strap:0x412b8ae2

enable spi-nand

ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 MHz)

Found recognized ID, rdid=0x00efaa21

init IP fail(0xffffffff)

init ddr ok

img sig ok

chksum ok

load img ok

s-boot

sec sig ok

decrypt img

jump 0xa0000000

SPI Nand ID=00efaa21

SPI Nand die chipsize=0x08000000 byte

SPI Nand dienum=1,

SPI Nand blocksize=0x00020000 byte,

SPI Nand pagesize=0x00000800 byte,

SPI Nand oobsize=0x00000040 byte,

[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024

[rtkn_scan_bbt, line 1822] block_v2r_num 00000400

[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,

[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd

INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb

load bbt v2r table:0 page:512

[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!

check v2r bbt table:0 OK

[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704

INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb

load bbt table:0 page:704

[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!

check bbt table:0 OK

[dump_BBT] Nand BBT Content

Congratulation!! No BBs in this Nand.

Realtek Crypto Engine v0.1

=>CPU Wake-up interrupt happen! GISR=09000084 

 

---Realtek RTL8197F boot code at 2020.07.14-20:40+0800 v3.4T-pre2.2 (993MHz)

Info: Load boot_info success!

== RTL8197 Aqara Gateway bootloader ==

boot_info: ver:0

kernel: newest:0, curr:0

rootfs: newest:0, curr:0

kernel[0]: sum:0x80bc, size:2233412, fail:0

      [1]: sum:0x808a, size:2233412, fail:0

rootfs[0]: sum:0x0000, size:9445444, fail:3

      [1]: sum:0x0000, size:9445444, fail:3

root_sum_check: off

watchdog_time: 0

boot_version: 1.0.0_0001

boot_magic: 0000917c

priv mode

Info: loading kernel 0 ... size 2233412

Info: checking kernel 0 ... 

Success!

Info: rootfs 0 is invalid

Info: rootfs 1 is invalid

Warn: no rootfs available.

---Ethernet init Okay!

<RealTek>
```

niceboygithub, 2021-10-21:
> kernel[0]: sum:0x80bc, size:2233412, fail:0
> 
>       [1]: sum:0x808a, size:2233412, fail:0
> 
> rootfs[0]: sum:0x0000, size:9445444, fail:3
> 
>       [1]: sum:0x0000, size:9445444, fail:3
> 

It is strange, I already use right method to calculate the sum value. It can not be 0x0000
Did you re-download AqaraGateway.exe to flash?

For quick fix, can you dump the boot_info in uboot?
```
NANDR 0xa0000 0xa0a00000 55
db 0xa0a00000 55
```

There is a [discussion ](https://community.home-assistant.io/t/downgrading-firmware-of-xiaomi-mijia-smart-multi-mode-gateway-zndmwg03lm/238938/3)about how to modify boot_info.

matteos1, 2021-12-17:
> I do not have signed key to sign firmware. Currently, it is not possible to have modified firmware for M2 EU.
> 
> To enable telnet in M2 EU, I just know one method to modify factory data to set "persist.app.tty_enable" to true.

could you explain this method?