niceboygithub/AqaraM1SM2fw #18: m2 eu

YanchikAmb, 2021-11-13:
Hi! Can I flash custom firmware to eu version hub m2? 
Do you know how enable telnet on hub m2 eu version?
How I can use Aqara Gateway integration via hub m2 eu version? 
Thanks.

niceboygithub, 2021-11-15:
M2 gateway EU version has signed firmware. Can not flash modified fw.

To enable telnet, there is a way with high risk. Not recommend that normal users to use it.
A simple solution is to buy CN version.

YanchikAmb, 2021-11-15:
oh I see, but in my country I can't buy zn version.
Thanks.

vladyslav-chupryn, 2022-02-02:
Hi! Could you please also upload the bootloader for the M2 EU version? I flashed my device with https://github.com/niceboygithub/AqaraM1SM2fw/blob/main/raw/bootloader_1.0.0_0001/bootload_1.0.0_0001.bin, and it stopped booting at all?

niceboygithub, 2022-02-10:
stay tuned for M2 EU hack

Nismonx, 2022-04-30:
Any advances on the M2 EU hack? My skills are limited but I'm more than happy to help out.

r3knit, 2022-05-19:
@niceboygithub > To enable telnet, there is a way with high risk.

Could you please share? I have got an almost bricked M2 EU flashed by mistake with modified fw :)

w1tw0lf, 2022-06-04:
Also keen on the modified firmware for M2 EU version.

closed 2022-08-31 by niceboygithub

niceboygithub, 2023-08-14:
Is anyone still interested on enabling telnet in M2 EU which can not flash modified firmware?

r3knit, 2023-08-14:
@niceboygithub 
> Is anyone still interested on enabling telnet in M2 EU which can not flash modified firmware?

Of course! Maybe i can find out how to use it even with its current firmware. Or mb there is a chance to flash it with something more interesting

niceboygithub, 2023-08-14:
Do you want to connect UART of M2?

I wrote a script that can clear the password of admin in M2.

If you wired the UART of M2, you shall use the script.

niceboygithub, 2023-08-14:
To flash firmware of M2 EU which use signed firmware, it is impossible because there is sign key to signed firmware.