niceboygithub/AqaraM1SM2fw #20: 3.3.4 Can't access telnet

STUkh, 2021-11-20:
Hey! Thanks for your hard work. I have Aqara M2 CN hub and successfully flashed 3.3.4 modded firmware through UART. But after couldn't get telnet access anyhow. 

Putty and other tools always immediately says that connection closed on remote side.

Also tried to connect by local address through direct connection, but also no luck.

This is my first attempt to get telnet access. Appreciate for any help.

[image]

STUkh, 2021-11-20:
Also I was able to connect to Serial connection (with baud speed 38400). Here's boot log:
```
uart ok
strap pin:0x412b8ae2
enable spi-nand
ROM ver:v1.1, sig:455cc27, time:2016.01.04-18:42+0800, CPU(400 MHz), DDR2(533 MHz)
load efuse ok
init IP ok
rom_progress: 0x0600006d
load_data_from_storage(260): 0xbfe01540, 0x00000000, 0xbfd16f44
load_data_from_spi_nand_flash(70): 0xbfe01540, 0x00000000, 0xbfe03e18
check_image_header(72): h(69,72,61,6d), s(69,72,61,6d)
img sig ok
rom_progress: 0x0c00006d
load_data_from_spi_nand_flash(81) 0x00000004 0x000024ba
load_data_from_spi_nand_flash(86): 0xbfe01d40, 0x00000001, 0xbfe03e18
load_data_from_spi_nand_flash(86): 0xbfe02540, 0x00000002, 0xbfe03e18
load_data_from_spi_nand_flash(86): 0xbfe02d40, 0x00000003, 0xbfe03e18
load_data_from_spi_nand_flash(86): 0xbfe03540, 0x00000004, 0xbfe03e18
load_data_from_spi_nand_flash(90) read done (size:9402)
chksum ok
rom_progress: 0x0e00006d
load img ok
rom_progress: 0x1000006d
jump 0xbfe01550

Booting...
SPI NAND clock not enable

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400
[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512
[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!
check v2r bbt table:0 OK
[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704
[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!
check bbt table:0 OK
[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
=>CPU Wake-up interrupt happen! GISR=89000084

---Realtek RTL8197F boot code at 2020.07.08-18:00+0800 v3.4T-pre2.1 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:1, curr:1
rootfs: newest:1, curr:1
kernel[0]: sum:0x81dc, size:2233348, fail:0
      [1]: sum:0x807d, size:2232324, fail:0
rootfs[0]: sum:0x0e37, size:9162756, fail:0
      [1]: sum:0xd6e1, size:9482244, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: loading kernel 1 ...  Done
Info: checking kernel 1 ... Success
Info: select rootfs 1
[... 180 lines cut ...]
hostname: Aqara-Hub-M2-5023
type: lumi.gateway.iragl5, model: AH_M2_BLE
Jan  1 00:00:09 mDNSResponder: mDNSResponder (Engineering Build) (May  7 2021 12:02:25) starting
Jan  1 00:00:09 mDNSResponder: Unable to parse DNS server list. Unicast DNS-SD unavailable
Jan  1 00:00:09 mDNSResponder: mDNSPlatformSourceAddrForDest: connect 1.1.1.1 failed errno 128 (Network is unreachable)
Jan  1 00:00:09 mDNSResponder: WARNING: mdnsd continuing as root becau[    9.690000] store_tty0_enable buf=disable
[    9.690000] , count=8
se user "nobody" does not exist
[   13.160000] i2c_designware i2c_designware.2: i2c_dw_handle_tx_abort: slave address not acknowledged (7bit mode)
[   13.220000] genirq: Flags mismatch irq 79. 00000083 (gpiolib) vs. 00000082 (wps btn)
[   16.100000] WlanSupportAbility = 0x3
[   16.110000] [ODM_software_init]
[   16.110000] [97F] Bonding Type 97FS, PKG1
[   16.110000] [97F] RFE type 0 PHY paratemters: DEFAULT
[   16.110000] clock 40MHz
[   16.110000] load efuse ok
[   16.110000] rom_progress: 0x200006f
[   16.110000] rom_progress: 0x400006f
[   16.190000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0] size
[   16.200000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0]
[   16.420000] [GetHwReg88XX][rtl8197Ffw]
[   16.430000] [GetHwReg88XX][rtl8197Ffw size]
[   16.870000] [97F] Default BB Swing=30

```

niceboygithub, 2021-11-21:
Strange. Looks fw was flashed by AqaraGateway but it still run old fw.
Can you flash again?

STUkh, 2021-11-21:
@niceboygithub thanks for response. Should I flash to `rootfs_1` also? Flashing to same place not effective. And one more thing - it is ok that 7/10 flashing attempts I get transfer error for file? Soldering looks good (not best, but good enough) and wires are 30cm long. Port USB 2.0 directly connected to MB with no sibling devices

STUkh, 2021-11-21:
@niceboygithub flash firmware to `rootfs_1` helped in my case 🙏 . After flashing I get prompted to login in Serial session the same as with telnet. It would be great to add to the docs the way with serial connection config (especially part with speed 38400). Also I think we don't need to get **curl** in `update.sh` script, it can be simplified a bit with only **wget** usage with `wget --no-proxy -O /tmp/m2_update.sh https://raw.githubusercontent.com/niceboygithub/AqaraM1SM2fw/main/modified/M2/m2_update.sh` and remove pretty big amount part of script fetching command

niceboygithub, 2021-11-22:
wget in buxybos is not supporting HTTPS

closed 2021-11-24 by STUkh