niceboygithub/AqaraM1SM2fw #22: How to retrieve lost data partition files?

z-x-f, 2021-12-14:
Hello, I have used the wrong tool (gateway3utils.exe) to flash the firmware of M2, which resulted in the loss of data partition files. Is there any way to restore the data partition?

rezmus, 2021-12-15:
if you didn't damaged mtd2 (factory) you can flash backup of someone else (best made after factory reset) and then factory reset hub (10x click). during factory reset only /data/musics/ is preserved (sounds). db is recreated from mtd2.

z-x-f, 2021-12-15:
Thank you for your answers, my mtd2 is not damaged.
@niceboygithub Could you please provide me with a backup of the data partition? Thank you very much.

niceboygithub, 2021-12-16:
I have uploaded the m2_musics.tar.gz.

You can use AqaraGateway.exe to flash the damaged parition then put the musics back as rezmus said.

z-x-f, 2021-12-16:
@niceboygithub Thank you for your help. This is my M2 Hub startup log, what should I do to fix it? 
<details>
<summary>m2.log</summary>
<pre>
uart ok
strap pin:0x412b8ae2
enable spi-nand
ROM ver:v1.1, sig:455cc27, time:2016.01.04-18:42+0800, CPU(400 MHz), DDR2(533 MHz)
load efuse ok
init IP ok
rom_progress: 0x0600006d
load_data_from_storage(260): 0xbfe01540, 0x00000000, 0xbfd16f44 
load_data_from_spi_nand_flash(70): 0xbfe01540, 0x00000000, 0xbfe03e18 
check_image_header(72): h(69,72,61,6d), s(69,72,61,6d) 
img sig ok
rom_progress: 0x0c00006d
load_data_from_spi_nand_flash(81) 0x00000004 0x000024ba 
load_data_from_spi_nand_flash(86): 0xbfe01d40, 0x00000001, 0xbfe03e18 
load_data_from_spi_nand_flash(86): 0xbfe02540, 0x00000002, 0xbfe03e18 
load_data_from_spi_nand_flash(86): 0xbfe02d40, 0x00000003, 0xbfe03e18 
load_data_from_spi_nand_flash(86): 0xbfe03540, 0x00000004, 0xbfe03e18 
load_data_from_spi_nand_flash(90) read done (size:9402) 
chksum ok
rom_progress: 0x0e00006d
load img ok
rom_progress: 0x1000006d
jump 0xbfe01550

Booting...
SPI NAND clock not enable

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400
[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512
[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!
check v2r bbt table:0 OK
[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704
[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!
check bbt table:0 OK
[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
=>CPU Wake-up interrupt happen! GISR=09000084 
 
---Realtek RTL8197F boot code at 2020.07.08-18:00+0800 v3.4T-pre2.1 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:0, curr:1
rootfs: newest:0, curr:0
kernel[0]: sum:0x80b7, size:2233348, fail:3
      [1]: sum:0x8030, size:2233348, fail:0
rootfs[0]: sum:0xa7a4, size:9537540, fail:0
      [1]: sum:0x0000, size:9568256, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: kernel 0 is invalid
[... 434 lines cut ...]
killall: zigbee_agent: no process killed
killall: ha_ble: no process killed
killall: miio_client: no process killed
killall: miio_agent: no process killed
killall: miio_client_helper_nomqtt.sh: no process killed
killall: mha_master: no process killed
killall: mha_basis: no process killed
killall: mzigbee_agent: no process killed
killall: mijia_automation: no process killed
killall: homekitserver: no process killed
mzigbee_agent has been killed
mha_basis has been killed
mijia_automation has been killed
miio_client has been killed
miio_agent has been killed
miio_client_helper_nomqtt.sh has been killed
mha_master has been killed
homekitserver has been killed
sh: write error: Device or resource busy
sh: write error: Device or resource busy
sh: write error: Device or resource busy
killall: zigbee_agent: no process killed
to_response, 0
</pre>
</details>

niceboygithub, 2021-12-16:
You can have a try with the following commands

```
ubidetach /dev/ubi_ctrl -m 8
ubiformat /dev/mtd8 -y
ubiattach /dev/ubi_ctrl -m 8 -d 1
ubimkvol /dev/ubi1 -N ubifs1 -m
mount -t ubifs -o sync ubi1_0 /data
```

I am not 100% sure that they work.

z-x-f, 2021-12-16:
It seems to work, but there are still errors. 
<details>
<summary>Log after executing the command</summary>
<pre>
uart ok
strap pin:0x416b8ae2
enable spi-nand
ROM ver:v1.1, sig:455cc27, time:2016.01.04-18:42+0800, CPU(400 MHz), DDR2(533 MHz)
load efuse ok
init IP ok
rom_progress: 0x0600006d
load_data_from_storage(260): 0xbfe01540, 0x00000000, 0xbfd16f44 
load_data_from_spi_nand_flash(70): 0xbfe01540, 0x00000000, 0xbfe03e18 
check_image_header(72): h(69,72,61,6d), s(69,72,61,6d) 
img sig ok
rom_progress: 0x0c00006d
load_data_from_spi_nand_flash(81) 0x00000004 0x000024ba 
load_data_from_spi_nand_flash(86): 0xbfe01d40, 0x00000001, 0xbfe03e18 
load_data_from_spi_nand_flash(86): 0xbfe02540, 0x00000002, 0xbfe03e18 
load_data_from_spi_nand_flash(86): 0xbfe02d40, 0x00000003, 0xbfe03e18 
load_data_from_spi_nand_flash(86): 0xbfe03540, 0x00000004, 0xbfe03e18 
load_data_from_spi_nand_flash(90) read done (size:9402) 
chksum ok
rom_progress: 0x0e00006d
load img ok
rom_progress: 0x1000006d
jump 0xbfe01550

Booting...
SPI NAND clock not enable

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400
[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512
[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!
check v2r bbt table:0 OK
[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704
[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!
check bbt table:0 OK
[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
=>CPU Wake-up interrupt happen! GISR=09000084 
Reboot Result from Watchdog Timeout!

---Realtek RTL8197F boot code at 2020.07.08-18:00+0800 v3.4T-pre2.1 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:0, curr:1
rootfs: newest:0, curr:0
kernel[0]: sum:0x80b7, size:2233348, fail:3
      [1]: sum:0x8030, size:2233348, fail:0
rootfs[0]: sum:0xa7a4, size:9537540, fail:0
      [1]: sum:0x0000, size:9568256, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
[... 555 lines cut ...]
to_response, 0
{"wifi_mac":"00E04C818686","set_zigbee_channel":"ok"}
cat: can't open '/data/factory/zigbee_device_lumi.0': No such file or directory
to_response, 1
udp send :[{"wifi_mac":"00E04C818686","cmd":"device_report","did":"lumi.0","model":"unknow","params":[{"res_name":"8.0.2157","value":"24388"}]}]
udp send encrypt data len:[144]
udp send failed! ret = -1
to_response, 0
{"wifi_mac":"00E04C818686","cmd":"device_report","did":"lumi.0","model":"unknow","params":[{"res_name":"8.0.2157","value":"24388"}]}
cat: can't open '/data/factory/zigbee_device_lumi.0': No such file or directory
to_response, 1
udp send :[{"wifi_mac":"00E04C818686","cmd":"device_report","did":"lumi.0","model":"unknow","params":[{"res_name":"8.0.2021","value":""}]}]
udp send encrypt data len:[128]
udp send failed! ret = -1
to_response, 0
{"wifi_mac":"00E04C818686","cmd":"device_report","did":"lumi.0","model":"unknow","params":[{"res_name":"8.0.2021","value":""}]}
cat: can't open '/data/factory/zigbee_device_lumi.0': No such file or directory
to_response, 1
udp send :[{"wifi_mac":"00E04C818686","cmd":"device_report","did":"lumi.0","model":"unknow","params":[{"res_name":"8.0.2021","value":""}]}]
udp send encrypt data len:[128]
udp send failed! ret = -1
to_response, 0
{"wifi_mac":"00E04C818686","cmd":"device_report","did":"lumi.0","model":"unknow","params":[{"res_name":"8.0.2021","value":""}]}
</pre>
</details>

niceboygithub, 2021-12-17:
it is in factory mode, you need to let it exit factory mode
```
get_factory_result = 0
```

please backup your factory data and properties
```
dd if=/dev/mtd2 of=/data/factory.bin
getprop
getprop persist.sys.hap_verifier
getprop persist.app.peerids_0
getprop persist.sys.coap_key
```

first, enter "exit" (I did remember the right command, you can enter "?" or "help" to query)

```
setprop persist.sys.factory_result true
```

then write back to factory data partition.

this command is dangerous if you properties are incorrected.  Suggest you enter "getprop" to get all properties then save them in case.
```
factory_data force_init
```

But I guest the values of factory data partition may corrupted.

z-x-f, 2021-12-17:
Wow, it was fixed, thank you so much.
There is another question, why does my kernel0 always fail:3
Even if the <code>sum</code> of <code>kernel 0</code> and <code>kernel 1</code> is the same, <code>kernel 1</code>  is fine, but <code>kernel 0</code> cannot be used.

closed 2021-12-17 by z-x-f