niceboygithub/AqaraM1SM2fw #23: New firmware 3.1.6_0022 for M1S

syssi, 2021-12-30:
I would be happy about a modified firmware flashable via `fw_update` for my lumi.gateway.acn01 (fw 3.1.6_0022).

niceboygithub, 2021-12-31:
WoW. The Great Syssi!!

Thanks for your comment.

I did not find firmware 3.1.6_0022 for M1S (acn01). How did you find it?
The version number of M1S firmware is already started from 3.3.x. 3.1.6 shall be very old version.

rezmus, 2021-12-31:
3.1.6 miio binaries, ota in mi home. aiot updates still come with 3.1.5 miio and i dont recommend 3.1.6 atm. saw ppl with issues after update which were resolved by downgrade to 3.1.5 (update in aqara app).

syssi, 2021-12-31:
My acn01 was delivered with firmware 3.0.4. I've used the Mi Home app to update the device to the advertised firmware version (3.1.6). I will follow @rezmus suggestion and try to downgrade the device.

syssi, 2021-12-31:
Hmmm.. the thing I still don't understand: Why is the firmware version of both OTAs old and the provided versions of your repository newer? How did you retrieve these new versions? Should I try one of the new versions instead of the downgrade?

rezmus, 2021-12-31:
m1s has separate binaries for aiot and miot. each ota has both. which one are important for you depends on app/mode you use. current version in aqara app is 3.3.6 aiot / 3.1.5 miot, and current version in mi home is 3.1.6 miot / 3.2.6 aiot. if you update in aqara app you will in fact downgrade miot version (3.1.6 > 3.1.5). i never updated my hub beyond 3.1.5, but saw a few ppl with different problems on 3.1.6, which were resolved after update/downgrade to 3.1.5 in aqara app. however if 3.1.6 works fine, you may just leave this version.

syssi, 2021-12-31:
My goal is to remove/drop all the cloud functionality. So I'm looking for a way to strip down the factory image. I want to use the device as remote controlled ZigBee *router* by flashing another zigbee firmware build. I've already found the GPIOs and serial interface to control the radio module.

Having a RGB led and some ALSA sound would be nice. At the moment I'm trying to understand how to control the RGB led without using `/bin/basis_cli`.

syssi, 2021-12-31:
@rezmus Thanks for the clarification. This helps a lot!

niceboygithub, 2021-12-31:
/dev/led is character device.
You may need to ioctl to control led.
Why you do not want to use basis_cli to control? 

M1S is very similar with to xiaomi gateway 3. AlexxIT and Vtel already implement flashing another zigbee firmware.
M1S shall be use the same way to flash. But M1S is very old device. M1S 2022 just released and more power than M1S.
I think that it is good challenge to try to flash another zigbee firmware or use M1S 2022's zigbee with Zigbee2MQTT/ZHA.

syssi, 2021-12-31:
Thanks for your support. I'm trying to recycle old & unused devices instead of buying something new. ;-)

syssi, 2022-01-02:
Downgrading the device via the Aqara Home app was easy.

```
$ getprop | grep fw
[ro.sys.fw_ver]: [3.3.4]
[ro.sys.mi_fw_ver]: [3.1.5]
```

syssi, 2022-01-02:
@niceboygithub Could you tell me how to modify/generate a custom rootfs.bin? I would like to drop some binaries and don't want to start some services on boot up.

rezmus, 2022-01-02:
don't remember for sure, but it was something like this

https://github.com/serrj-sv/lumi.gateway.mgl03/blob/main/tools/rtk-tools/cvimg.c

cc cvimg.c -o cvimg -DCONFIG_RTL_819XD

sudo unsquashfs root.bin

sudo mksquashfs <ROOT_DIR> root.squashfs -noappend -always-use-fragments -comp xz -no-xattrs

(append 0x00 0x00) to the end of root.squashfs)

./cvimg root root.squashfs <new_root.img> 0x2d0000 0xe00000

take stock root.bin, strip first 16 and last 4 bytes and follow above. without unsquashfs/mksquashfs you should get exactly same file.

niceboygithub, 2022-01-02:
Another method. Just modify post_init.sh

niceboygithub, 2022-01-03:
You can also reference to [this](https://github.com/niceboygithub/XiaomiGateway3fw/blob/d36e9376db199694edcbcfe6d53de01e7399cd33/scripts/gateway3utils.py#L260)

(Please forget my poor python skill)

syssi, 2022-01-04:
Thanks for your support. I've to admit I give up at this point because I realized the zigbee radio at my M1S is a JN5189 and no JN5169. It looks like the JennicModuleProgrammer doesn't have JN5189 support and [this](https://github.com/igo-r/Lumi-Router-JN5169) router firmware is JN5169 specific.

closed 2022-01-04 by syssi