niceboygithub/AqaraM1SM2fw #25: Please upload bootloader for the M2 EU version

vladyslav-chupryn, 2022-02-08:
Hi! Could you please also upload the bootloader for the M2 EU version? I flashed my device with https://github.com/niceboygithub/AqaraM1SM2fw/blob/main/raw/bootloader_1.0.0_0001/bootload_1.0.0_0001.bin, and it stopped booting at all?

Log from PuTTY:
uart ok
strap:0x412b8ae2
enable spi-nand
ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 M Hz)

Found recognized ID, rdid=0x00efaa21
init IP fail(0xffffffff)
init ddr ok
img sig err
ECC_DECODE_FAIL,ecc_start_addr = 0xbfe006e8
img sig err
ECC_DECODE_FAIL,ecc_start_addr = 0xbfe006e8
img sig err
ECC_DECODE_FAIL,ecc_start_addr = 0xbfe006e8
img sig err
load img fail(0xffffffff)
s-boot
sec sig err
fail

rezmus, 2022-02-09:
ask on https://t.me/xiaomi_gw_v3_hack

rezmus, 2022-02-09:
https://ufile.io/vidnpnot

vladyslav-chupryn, 2022-02-10:
Hi! It is booting now, thanks a lot!!!
Could you please also upload boot_info by any chance?

Now I am getting the following log via PuTTY:
uart ok
strap:0x412b8ae2
enable spi-nand
ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 M                                                              Hz)

Found recognized ID, rdid=0x00efaa21
init IP fail(0xffffffff)
init ddr ok
img sig ok
chksum ok
load img ok
s-boot
sec sig ok
decrypt img
jump 0xa0000000

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400
[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512
[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!
check v2r bbt table:0 OK
[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704
[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!
check bbt table:0 OK
[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
Realtek Crypto Engine v0.1
=>CPU Wake-up interrupt happen! GISR=09000084

---Realtek RTL8197F boot code at 2020.07.14-20:40+0800 v3.4T-pre2.2 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:1, curr:1
rootfs: newest:0, curr:1
kernel[0]: sum:0x81ab, size:2233348, fail:3
      [1]: sum:0x80bc, size:2233412, fail:3
rootfs[0]: sum:0x64e3, size:9601028, fail:3
      [1]: sum:0xfbb5, size:9445444, fail:3
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: kernel 1 is invalid
Info: kernel 0 is invalid
Warn: all kernels are invalid !

---Ethernet init Okay!
<RealTek>

I tried to flash both rootfs and linuxfs from  https://github.com/niceboygithub/AqaraM1SM2fw/tree/main/original/M2_EU/3.3.0_0021_0526 with the flag "Signed firmware" via aqaragateway.exe, but seems like there is an issue with incorrect signature.

niceboygithub, 2022-02-10:
boot_info depends on the kernel and rootfs.
If you use aqaragateway.exe to flash firmware, it will calculate the checksum of kernel and rootfs then write to boot_info and reset the number of booting failure.

vladyslav-chupryn, 2022-02-10:
Thanks a lot for your answer!

vladyslav-chupryn, 2022-02-14:
Hi! Could you please upload a newer version of firmware for M2 EU?

Unfortunately, I am facing continuous reboots after flashing 3.3. Log is the following:
uart ok
strap:0x412b8ae2
enable spi-nand
ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 MHz)

Found recognized ID, rdid=0x00efaa21
init IP fail(0xffffffff)
init ddr ok
img sig ok
chksum ok
load img ok
s-boot
sec sig ok
decrypt img
jump 0xa0000000

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400
[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512
[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!
check v2r bbt table:0 OK
[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704
[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!
check bbt table:0 OK
[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
Realtek Crypto Engine v0.1
=>CPU Wake-up interrupt happen! GISR=09000084

---Realtek RTL8197F boot code at 2020.07.14-20:40+0800 v3.4T-pre2.2 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:0, curr:0
rootfs: newest:0, curr:1
kernel[0]: sum:0x80bc, size:2233412, fail:0
      [1]: sum:0x80bc, size:2233412, fail:2
rootfs[0]: sum:0xfba6, size:9445444, fail:3
      [1]: sum:0xfba6, size:9445444, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: loading kernel 0 ... size 2233412
Info: checking kernel 0 ...
Success!
Info: rootfs 0 is invalid
Info: loading rootfs 1 ...  Done
Info: checking rootfs 1 ... Info: checking rootfs 1 ...
rootfs check Success
Info: select rootfs 1
cmdline:root=/dev/mtdblock7 console=ttyS0,38400
Info: booting...
Jump to image start=0x80a00000...
decompressing kernel:
[... 187 lines cut ...]
[   16.140000] [97F] RFE type 0 PHY paratemters: DEFAULT
[   16.140000] clock 40MHz
[   16.140000] load efuse ok
[   16.140000] rom_progress: 0x200006f
[   16.140000] rom_progress: 0x400006f
[   16.210000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0] size
[   16.230000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0]
[   16.450000] [GetHwReg88XX][rtl8197Ffw]
[   16.460000] [GetHwReg88XX][rtl8197Ffw size]
[   16.900000] [97F] Default BB Swing=30
uart ok
strap:0x412b8ae2
enable spi-nand
ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 MHz)

Found recognized ID, rdid=0x00efaa21
init IP fail(0xffffffff)
init ddr ok
img sig ok
chksum ok
load img ok
s-boot
sec sig ok
decrypt img
jump 0xa0000000

niceboygithub, 2022-02-25:
I uploaded, but I am not sure that what it can help for you.

vladyslav-chupryn, 2022-02-28:
Thanks a lot! I will try.

closed 2022-08-31 by niceboygithub