niceboygithub/AqaraM1SM2fw #29: FW needed 3.3.9_0005.0616 for M2

gnep88, 2022-03-04:
Hi @niceboygithub 

Would you please upload the new FW 3.3.9_0005.0616 for M2? Thanks!

niceboygithub, 2022-03-07:
I did not get the update yet.

gnep88, 2022-03-07:
This is the link I captured last week…

But I do not know why it is not available any more. Do you have any insights why it suddenly disappears from my App ? Thanks 

https:\/\/cdn.aqara.com\/cdn\/opencloud-product\/mainland\/product-firmware\/prd\/lumi.gateway.iragl5\/20220228154507_lumi.gateway.iragl5_AIOT_V3.3.9_0005.0616_20220223_613d45.bin

niceboygithub, 2022-03-09:
That’s meaning that it is a buggy version. Let’s wait for the next version.

gnep88, 2022-03-15:
@niceboygithub thanks. A new version is just made available. Please check V3.3.9_0013.0616

https:\/\/cdn.aqara.com\/cdn\/opencloud-product\/mainland\/product-firmware\/prd\/lumi.gateway.iragl5\/20220309160737_lumi.gateway.iragl5_AIOT_V3.3.9_0013.0616_20220309_500f57.bin

closed 2022-03-24 by gnep88