niceboygithub/AqaraM1SM2fw #30: Stuck at "Downloading the flasher."

wr, 2022-03-24:
Have an M2 hub (stock firmware) and trying to flash 3.3.9_0013.0616_modified.bin using aqaragateway.exe

It shows a progress bar for "Downloading the flasher" (and TX lights up on the TTY USB) but once the progress bar fills up it hangs.

```
Do NOT power up your gateway.
IF it is already power up, remove power cable!

Generating padded firmware, please wait...!
Please power up gateway!
If your gateway is powered up, disconnect usb cable and reconnect it.
Downloading the flasher.
```

omgseb, 2022-03-29:
+1, same issue here. US M2 hub, stuck at "downloading the flasher".

Nismonx, 2022-04-28:
same issue with rootfs_3.4.0_0004.0616_modified.bin on EU model

Nismonx, 2022-04-28:
log from putty after manual restart:

uart ok
strap:0x412b8ae2
enable spi-nand
ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 M                                                                                        Hz)

No recognized ID, rdid=0x00a1e47f
init IP fail(0xffffffff)
init ddr ok
img sig ok
chksum ok
load img ok
s-boot
sec sig ok
decrypt img
jump 0xa0000000

SPI Nand ID=0000a1e4
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400
[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512
[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!
check v2r bbt table:0 OK
[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704
[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!
check bbt table:0 OK
[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
Realtek Crypto Engine v0.1
=>CPU Wake-up interrupt happen! GISR=09000084

---Realtek RTL8197F boot code at 2021.05.13-10:48+0800 v3.4T-pre2.2 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:1, curr:1
rootfs: newest:1, curr:1
kernel[0]: sum:0x8062, size:2233412, fail:0
      [1]: sum:0x8016, size:2233412, fail:0
rootfs[0]: sum:0xf2be, size:9719876, fail:0
      [1]: sum:0xee7c, size:9723972, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: loading kernel 1 ... size 2233412
Info: checking kernel 1 ...
Success!
Info: loading rootfs 1 ...  Done
Info: checking rootfs 1 ... Info: checking rootfs 1 ...
rootfs check Success
Info: select rootfs 1
cmdline:root=/dev/mtdblock7 console=ttyS0,38400
Info: booting...
Jump to image start=0x80a00000...
decompressing kernel:
Uncompressing Linux... done, booting the kernel.
done decompressing kernel.
[... 170 lines cut ...]
fw_manager.sh revision: 2
pre-revision: 2

hostname: Aqara-Hub-M2-00EC
type: lumi.gateway.iragl7, model: AH_M2_BLE
Jan  1 00:00:09 mDNSResponder: mDNSResponder (Engineering Build) (May 12 2021 15:43:36) starting
Jan  1 00:00:09 mDNSResponder: Unable to parse DNS server list. Unicast DNS-SD unavailable
Jan  1 00:00:09 mDNSResponder: mDNSPlatformSourceAddrForDest: connect 1.1.1.1 failed errno 128 (Network is unreachable)
Jan  1 00:00:09 mDNSResponder: WARNING: mdnsd continuing as root because user "nobody" does not exist
[    9.660000] store_tty0_enable buf=disable
[    9.660000] , count=8
[   14.710000] genirq: Flags mismatch irq 79. 00000083 (gpiolib) vs. 00000082 (wps btn)
[   16.370000] WlanSupportAbility = 0x3
[   16.390000] [ODM_software_init]
[   16.390000] [97F] Bonding Type 97FS, PKG1
[   16.390000] [97F] RFE type 0 PHY paratemters: DEFAULT
[   16.390000] clock 40MHz
[   16.390000] load efuse ok
[   16.390000] rom_progress: 0x200006f
[   16.390000] rom_progress: 0x400006f
[   16.460000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0] size
[   16.480000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0]
[   16.690000] [GetHwReg88XX][rtl8197Ffw]
[   16.700000] [GetHwReg88XX][rtl8197Ffw size]
[   17.150000] [97F] Default BB Swing=30

Nismonx, 2022-04-28:
I've hooked up a virtual com splitter to capture this:

strap:0x412b8ae2
enable spi-nand
ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 M                                                                                        Hz)
console
<RealTek>uuuuu
Unknown cmd
<RealTek>
<RealTek>dbgmsg 3
dbg level:0x00000003
<RealTek>ri 0 1 1
load efuse ok
rom:0x0200006d
0xb8000780: 0x00028e8f
0xb8000784: 0x00000000
0xb8000788: 0x00000000
0xb800078c: 0x00000000
0xb8000790: 0x00000000
0xb8000794: 0x00000000
0xb8000798: 0x00000000
0xb800079c: 0x00000000
0xb80007a0: 0x00000000
0xb80007a4: 0x00000000
0xb80007a8: 0xb139f925
0xb80007ac: 0x773c16bf
0xb80007b0: 0x2c83da67
0xb80007b4: 0x380c40c6
0xb80007b8: 0x2e00a4dd
0xb80007bc: 0xd27fbd02
0xb80007c0: 0x3ee4c655
0xb80007c4: 0x0f2b545c
0xb80007c8: 0xdc3d6617
0xb80007cc: 0x4a473292
0xb80007d0: 0x00000000
0xb80007d4: 0x00000000
0xb80007d8: 0xb537693b
0xb80007dc: 0x02706064
0xb80007e0: 0xb0feac56
0xb80007e4: 0x8c86b3c3
0xb80007e8: 0x171a3a85
0xb80007ec: 0xd106623c
0xb80007f0: 0x02000103
0xb80007f4: 0x9cb63a2f
0xb80007f8: 0x00000005
0xb80007fc: 0x00000000
loc:0x00000220,offset:0x00000044,word_enb:0x0000000e
loc:0x00000220,init_dev:0x00000002
0x220   02 00 00 00 00 00 00 00
read eFuse 0x005 byte
ctr: 0x00
ctr2: 0x00
ctr3: 0x00
ctr4: 0x00
exec sram: 0x00000000
boot adr: 0x00000000
img off: 0x00000000
dev: 0x00000002
hp adr: 0x00000000
hp len: 0x00000000
er st adr: 0x00000000
rom:0x0400006d
ss_init_rxi310
ms_ctrl_0_map=0xb8142000
bond:0x0000000a
MCM 64/32MB

 dram_init_clk_frequency ,ddr_freq=1066 (Mbps), 533 (MHZ)
mck_ck=7, 1112
dev_map=0xb8142000
[... 171 lines cut ...]
memcpy 4 0xa0400100 passed

memcpy 8 0xa0400100 passed

memcpy 12 0xa0400100 passed

memcpy 16 0xa0400100 passed

memcpy 20 0xa0400100 passed

memcpy 24 0xa0400100 passed

memcpy 28 0xa0400100 passed
 memcpy_test done
init ddr ok
rom:0x0800006d
<RealTek>xmrx 0xa0000000
recv ok len:0x0001ce80
<RealTek>j a0000000
Jump to 0xA0000000
ۼ▒%▒m$)▒<)4z▒)▒▒▒6▒▒;y▒▒((▒▒▒Xഖ,▒▒▒(l▒ۼ▒%▒m$)▒<+4z▒)▒▒

then when "downloading the flasher" bar is done, nothing happens and get the above artefact.

this is as far I can log things.

niceboygithub, 2022-04-29:
do not flash customize fw to EU version.
The fw of EU version is signed. If you flash customized fw to gateway, it will become brick.

Nismonx, 2022-04-30:
@niceboygithub, thanks for the heads up. I genuinely thought it would be fine.

closed 2022-07-27 by niceboygithub