niceboygithub/AqaraM1SM2fw #32: Add lumi.gateway.aeu01 firmware. Please!

lopatin777, 2022-04-09:
Please help! 
Need original firmware for lumi.gateway.aeu01.

niceboygithub, 2022-04-11:
I do no have such device on hand. It is hard to me to provide original firmware. Maybe someone else can help.

lopatin777, 2022-04-11:
it’s very sadly.

maikthulhu, 2022-05-05:
I have one of these that identifies as lumi.gateway.aeu01. I'm in the US but picked it up off of Amazon. I also probably have the skills necessary to crack the case and dump the firmware.

@niceboygithub Would it be helpful if I dumped it and did a PR so you can have the image?

martyanov1990, 2022-06-27:
Hello. Has the firmware for the m1s eu not arrived?

Otnow, 2022-07-09:
```json
"releaseDate": 1656555792000,
"part": "main",
"downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.aeu01/20220622151632_lumi.gateway.aeu01_AIOT_V3.4.2_0016.0616_20220622_4923c0.bin",
"upgradeTakesTime": 40,
"updateLog": "1.Optimize device related functions;\n2.Fix known issues;",
"updateTime": 1656555792000,
"isNecessary": 0,
"fileSize": 13496424,
"createTime": 1656418454000,
"firmwareMD5": "aacdef9333aee3e5b496c1c3584923c0",
"state": 3,
"isForce": false,
"firmwareVersion": "3.4.2_0016.0616",
"imageType": "bin",
"firmwareCRC32": "5DCCB0E4"
```

closed 2022-07-10 by lopatin777

sorryusernameisalreadytaken, 2022-07-18:
> ```json
> "releaseDate": 1656555792000,
> "part": "main",
> "downloadUrl": [...]

@Otnow can you help us here getting more details about your trick to get the direct dl links?
https://github.com/Koenkk/zigbee2mqtt/discussions/13193

Otnow, 2022-07-18:
@sorryusernameisalreadytaken, answered in discussion https://github.com/Koenkk/zigbee2mqtt/discussions/13193.