niceboygithub/AqaraM1SM2fw #34: use scripts update m1s fw shows Update failed, reason: getpack failed!

casatift, 2022-04-15:
use scripts update m1s fw shows Update failed, reason: getpack failed!

> # chmod a+x /tmp/m1s_update.sh && /tmp/m1s_update.sh
m1s_update.sh revision: 2
type: lumi.gateway.acn01, model: AH_M1S
platform: aiot, path: /tmp, sign: 0
Unpack path : /tmp/ota_dir
Firmware path : /data/ota_dir
Available ROM size(b): 26251264
Available RAM size(b): 22396928
Get packages, please wait...
Update failed, reason: getpack failed!

mikeinblue, 2022-04-18:
The same problem.

niceboygithub, 2022-04-29:
I fixed it. Please try again.

vinter-wu, 2022-05-11:
# chmod a+x /tmp/m1s_update.sh && /tmp/m1s_update.sh
m1s_update.sh revision: 2
type: lumi.gateway.acn01, model: AH_M1S
platform: miot, path: /tmp, sign: 0
Unpack path : /tmp/ota_dir
Firmware path : /data/ota_dir
Available ROM size(b): 26394624
Available RAM size(b): 15499264
Get packages, please wait...
md5sum: can't open '/tmp/rootfs.bin': No such file or directory
Update failed, reason: getpack failed!

closed 2022-08-02 by niceboygithub