niceboygithub/AqaraM1SM2fw #38: Aqara P3 Firmware Update

seanzhang98, 2022-05-19:
Aqara released the 3.4.0_004.0616 firmware for P3 with new device support. Please release a modified version. Thanks.

niceboygithub, 2022-06-21:
Can you provide the url?

seanzhang98, 2022-07-02:
@niceboygithub how can I obtain the URL? any tutorial?

Otnow, 2022-07-02:
> @niceboygithub how can I obtain the URL? any tutorial?

Try this guide: https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

SaintWe, 2022-07-04:
```
{
        "releaseDate": 1656556133000,
        "part": "main",
        "downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.aircondition.acn05/20220622155040_lumi.aircondition.acn05_AIOT_V3.4.2_0016.0616_20220622_37358c.bin",
        "upgradeTakesTime": 0,
        "updateLog": "1.Optimize device related functions;\n2.Fix known issues;",
        "updateTime": 1656556133000,
        "isNecessary": 0,
        "fileSize": 13731944,
        "createTime": 1656327514000,
        "firmwareMD5": "8d25d86e088c2a92e16893a3a237358c",
        "state": 3,
        "isForce": false,
        "firmwareVersion": "3.4.2_0016.0616",
        "imageType": "bin",
        "firmwareCRC32": "4C69F94D"
}
```

closed 2022-07-06 by seanzhang98