niceboygithub/AqaraM1SM2fw #41: Firmware Region

aminhusni, 2022-06-28:
Hi, 

I have an Aqara M1S CN region. I am wondering if flashing the original firmware provided here will change it to use US/Others region when using the Aqara Home app. 
So far I had to deal with slow CN servers :(

rezmus, 2022-06-29:
get root, change model to lumi.gateway.aeu01 and connect to global region (dont update hub after).

aminhusni, 2022-07-05:
> get root (in mijia mode without need to solder), change model to lumi.gateway.aeu01 and connect to global region (dont update hub after).

Which file specifically do I need to change?

aminhusni, 2022-07-05:
>  lock it with chattr (custom busybox)

Do you have a guide on how to proceed with this last part about locking it with Chattr?
I got everything done up to that point but I can't switch to Aqara mode (aiot mode?) without resetting everything and locking back telnet into factory state.

rezmus, 2022-07-05:
post_init.sh call was removed in last aiot fw 3.4.2 so it will only work if you did not update in aqara app in last 2 weeks and have fw 3.4.0 or lower (check ro.sys.fw_ver in /etc/build.prop).

check https://github.com/niceboygithub/AqaraGateway

follow "Not Flash modified firmware method (NOT for G2H, E1 hub, G3)" section. you can skip mosquitto download.

it will install curl and post_init.sh. before reboot **make sure script is OK** same as on repo. after reboot you should have telnet access without need to send miio command.

now you can download busybox and lock post_init.sh with chattr cmd

```
/data/bin/curl -s -k -L -o /data/bin/busybox https://busybox.net/downloads/binaries/1.21.1/busybox-mipsel
chmod +x /data/bin/busybox
/data/bin/busybox chattr +i /data/scripts/post_init.sh
```

after you switch mode to aiot telnet access should remain so last thing to do is gw model change and wifi reset.

aminhusni, 2022-07-05:
Thank you @rezmus for the excellent help! Works like a charm :)

closed 2022-07-05 by aminhusni