niceboygithub/AqaraM1SM2fw #44: M1S EU version

martyanov1990, 2022-07-10:
Hello. Can you tell me, support for the Eu version is planned? Before version 3.4.2_0008, everything worked. Accidentally updated and it does not work.

martyanov1990, 2022-07-11:
> connect to xiaomi app for solderless root and downgrade or use via homekit controller.

In this case the aqara n100 zigbee does not work.

rezmus, 2022-07-11:
you can go back to aiot after downgrade. in 3.4.0+ they removed root entry point for aiot.

martyanov1990, 2022-07-11:
> https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.aeu01/20220329120721_lumi.gateway.aeu01_AIOT_V3.4.0_0004.0616_20220328_3be1ed.bin
> 
> you can extract rootfs/kernel with 7zip or strip 0x68 bytes header and extract with tar. make sure your hub is lumi.gateway.aeu01 and you know what you are doing if you don't want to solder or worse.

Can you write how to flash it? I know how to solder, maybe through uart and Aqaragateway.exe will be easier?

rezmus, 2022-07-11:
download firmware to /data/ with curl. make sure you have enough empty space and check md5sum before. should be X.

martyanov1990, 2022-07-11:
> before you do something it is always recommended to do full backup and send with nc/ftp.
> 
> you can use http (not https) to download with wget directly. check with df before how much space you have. you may need ~26mb for firmware and tar extraction.
> 
> ```
> wget -O /data/downgrade.bin http://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.aeu01/20220329120721_lumi.gateway.aeu01_AIOT_V3.4.0_0004.0616_20220328_3be1ed.bin
> fw_manager.sh -u -n /data/downgrade.bin
> ```

[image]

rezmus, 2022-07-11:
```
mv /data/downgrade.bin /tmp/
fw_manager.sh -u -n /tmp/downgrade.bin
```

martyanov1990, 2022-07-11:
> ```
> mv /data/downgrade.bin /tmp/
> fw_manager.sh -u -n /tmp/downgrade.bin
> ```

Still doesn't want to.
[image]

rezmus, 2022-07-11:
maybe it is better to unpack/flash only rootfs not whole firmware (less space needed). wait for @niceboygithub he will guide you.

niceboygithub, 2022-07-11:
download firmware bin to /tmp directly.

/data/bin/curl -s -k -L -o /tmp/downgrade.bin https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.aeu01/20220329120721_lumi.gateway.aeu01_AIOT_V3.4.0_0004.0616_20220328_3be1ed.bin;

rezmus, 2022-07-11:
hub is in miot mode so fw expect miot ota and runs fw_miot_unpack, not fw_unpack.

martyanov1990, 2022-07-12:
> The fw of M1S eu is signed. The update command is “fw_manager.sh -u -s downgrade.bin”
> 
> sorry for forgot that.

Now it swears at the signature.
[image]

niceboygithub, 2022-07-12:
Can you try this command?

```
setprop persist.sys.cloud aiot
fw_manager.sh -u -n /tmp/downgrade.bin
```

martyanov1990, 2022-07-12:
It worked. But I can't lock the files. On chattr +i /data/scripts/post_init.sh -sh: chattr: not found.

rezmus, 2022-07-12:
https://github.com/niceboygithub/AqaraM1SM2fw/issues/41#issuecomment-1175250774

martyanov1990, 2022-07-12:
Great! Everything works. Thank you all very much. I will now write instructions for future generations and close the issues.

martyanov1990, 2022-07-12:
1. Switch to xiaomi home mode in 3.4.2
2. use [soft_hack](https://gist.github.com/zvldz/1bd6b21539f84339c218f9427e022709) to enable telnet
3. downgrade the fw to 3.4.0.
	3.1 /data/bin/curl -s -k -L -o /tmp/downgrade.bin https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.aeu01/20220329120721_lumi.gateway.aeu01_AIOT_V3.4.0_0004.0616_20220328_3be1ed.bin
	3.2 setprop persist.sys.cloud aiot
	3.3 fw_manager.sh -u -n /tmp/downgrade.bin
4. add to HA by AqaraGateway
5. Modify the firmware. 
	5.1 mkdir /data/bin
	5.2 cd /data/bin
	5.3 wget -O /data/bin/curl "http://master.dl.sourceforge.net/project/mgl03/bin/curl?viasf=1"; chmod +x /data/bin/curl
	5.4 /data/bin/curl -s -k -L -o /data/bin/mosquitto https://raw.githubusercontent.com/niceboygithub/AqaraM1SM2fw/main/binutils/mosquitto; chmod a+x /data/bin/mosquitto
	5.5 mkdir /data/scripts
	5.6 cd /data/scripts
	5.7 /data/bin/curl -s -k -L -o /data/scripts/post_init.sh https://raw.githubusercontent.com/niceboygithub/AqaraM1SM2fw/main/binutils/post_init.sh
	5.8 chmod +x /data/scripts/post_init.sh
6. telnet to gateway and lock post_init.sh by using the following command
	6.1 mkdir /data/bin
	6.2 cd /data/bin
	6.3 wget -O /data/bin/curl "http://master.dl.sourceforge.net/project/mgl03/bin/curl?viasf=1"; chmod +x /data/bin/curl
	6.4 /data/bin/curl -s -k -L -o /data/bin/busybox https://busybox.net/downloads/binaries/1.21.1/busybox-mipsel
	6.5 chmod +x /data/bin/busybox
	6.6 /data/bin/busybox chattr +i /data/scripts/post_init.sh
7. switch to Aqara Home Mode.
8.  lock the firmware update by using the following command
	8.1 mkdir /data/bin
	8.2 cd /data/bin
	8.3 wget -O /data/bin/curl "http://master.dl.sourceforge.net/project/mgl03/bin/curl?viasf=1"; chmod +x /data/bin/curl
	8.4 /data/bin/curl -s -k -L -o /data/ota_dir/busybox https://busybox.net/downloads/binaries/1.21.1/busybox-mipsel; chmod +x /data/ota_dir/busybox
	8.5 mkdir -p /data/ota_dir
	8.6 touch /data/ota_dir/lumi_fw.tar
	8.7 /data/ota_dir/busybox chattr +i /data/ota_dir/lumi_fw.tar

closed 2022-07-12 by martyanov1990