niceboygithub/AqaraM1SM2fw #45: M2 stop on "Downloading the flasher"

KFlab, 2022-07-21:
我用的是 2022 04 生产的 M2网关 
设备固件是 3.2.4_0013.0520
没有焊线 用手按的杜邦线
开机可以在串口看到回显

固件升级时候红灯常亮 然后等好久软件都没有进度
"Downloading the flasher" 一步 然后进度条读完
后面就没有然后了

是不是因为手按接触不良？我已经买了烧录夹 这两天快递到了我试试

niceboygithub, 2022-07-21:
Please use fixture.

niceboygithub, 2022-07-21:
> M1S网关 用米家模式更新之后是不是可以切回aqara模式使用？是不是我就不用非得烧录M2了

M1S (not 2022) can use Mi home mode to open telnet then flash cfw

KFlab, 2022-07-21:
GOT! thanks , I'll try fixture.

KFlab, 2022-07-22:
[ReceivedTofile-TCPCLIENT-2022_7_22_15-05-19.log](https://github.com/niceboygithub/AqaraM1SM2fw/files/9165612/ReceivedTofile-TCPCLIENT-2022_7_22_15-05-19.log)

KFlab, 2022-07-22:
[untitle.txt](https://github.com/niceboygithub/AqaraM1SM2fw/files/9166268/untitle.txt)
I use Monitor get some COM logs here

niceboygithub, 2022-07-22:
The pc (tx) to m2 (rx) is not connected well. So the m2 can not enter download mode.

KFlab, 2022-07-23:
I notice my M2 type is [ ZHWG17LM ]  not 12 is that important?

joelliang, 2022-07-23:
I just bought a M2.
I have the same issue. Stop at "Downloading the flasher". And I saw many open issues related to that.

I have a solid connection (I soldered three jumper cable to the PCB of M2)
And I can confirm by use Putty, after boot up, it can echo what ever I type.

My M2 type is [ZHWG17LM] as well.

Is there another way to flash the modified fw into M2?

joelliang, 2022-07-23:
> Is there another way to flash the modified fw into M2?

I'm asking this because I saw some guy more than one year ago, use the gateway3utils to flash M2 firmware.
https://bbs.hassbian.com/forum.php?mod=redirect&goto=findpost&ptid=12137&pid=361014

Or, is there any thing I can help with, to diagnostic the issue.

joelliang, 2022-07-23:
I use a Serial Port Splitter to get some log when using the aqaragateway.exe to flash the M2
[flashing_m2.log](https://github.com/niceboygithub/AqaraM1SM2fw/files/9173541/flashing_m2.log)

At first, the aqaragateway.exe open the serial port as 38400,N,8,1
After showing the "Downloading the flasher", and the progress bar become 100%, it open the serial port as 230400,N,8,1

niceboygithub, 2022-07-23:
Can you put serial port to 230400 and get the logs?

KFlab, 2022-07-23:
> [untitle.txt](https://github.com/niceboygithub/AqaraM1SM2fw/files/9166268/untitle.txt) I use Monitor get some COM logs here

all logs before 230400 are all TIMEOUT 
I use software "Accessport" get those logs
software change frequency automatically

joelliang, 2022-07-24:
> Can you put serial port to 230400 and get the logs?

I could not capture those logs. The Putty return logs for 38400 from the virtual serial port, even though I open the port as 230400. And return garbage data for the 230400 period. Maybe a bug in the serial port splitter software.

But I think, it should be the same as @KFlab captured.
From KFlab's logs, it got two bytes from M2, and all subsequent data writing to M2 are timeout.
```
2057	16:53:55.280	0.00330400	aqaragateway.e	IRP_MJ_CREATE                       	COM3	SUCCESS	Port Opened	
2058	16:53:55.283	0.00000550	aqaragateway.e	IOCTL_SERIAL_SET_QUEUE_SIZE         	COM3	SUCCESS	InSize: 4096, OutSize: 4096	
2061	16:53:55.283	0.00126290	aqaragateway.e	IOCTL_SERIAL_SET_BAUD_RATE          	COM3	SUCCESS	Baud Rate: 230400	
2062	16:53:55.284	0.00028750	aqaragateway.e	IOCTL_SERIAL_SET_RTS                	COM3	SUCCESS		
2063	16:53:55.285	0.00053160	aqaragateway.e	IOCTL_SERIAL_SET_DTR                	COM3	SUCCESS		
2064	16:53:55.285	0.00053450	aqaragateway.e	IOCTL_SERIAL_SET_LINE_CONTROL       	COM3	SUCCESS	StopBits: 1, Parity: No, DataBits: 8	
2065	16:53:55.286	0.00036740	aqaragateway.e	IOCTL_SERIAL_SET_CHARS              	COM3	SUCCESS	EofChar: 0x0, ErrorChar: 0x0, BreakChar: 0x0, EventChar: 0x0, XonChar: 0x11, XoffChar: 0x13	
2066	16:53:55.286	0.00050090	aqaragateway.e	IOCTL_SERIAL_SET_HANDFLOW           	COM3	SUCCESS	ControlHandShake: 0x1, FlowReplace: 0x40, XonLimit: 32768, XoffLimit: 8192	
2067	16:53:55.287	0.05112310	aqaragateway.e	IOCTL_SERIAL_PURGE                  	COM3	SUCCESS	Purge: TXABORT RXABORT TXCLEAR RXCLEAR	
2068	16:53:55.338	0.00006280	aqaragateway.e	IRP_MJ_WRITE                        	COM3	SUCCESS	Length: 2, Data: ..	
2069	16:53:55.338	3.01064930	aqaragateway.e	IRP_MJ_READ                         	COM3	TIMEOUT	Length: 0, Data: 	
2070	16:53:58.349	3.00216010	aqaragateway.e	IRP_MJ_READ                         	COM3	TIMEOUT	Length: 0, Data: 	
2071	16:54:01.351	3.00572260	aqaragateway.e	IRP_MJ_READ                         	COM3	TIMEOUT	Length: 0, Data: 	
```

Is that possible to modify the [gateway3utils.py](https://github.com/niceboygithub/XiaomiGateway3fw/blob/master/scripts/gateway3utils.py) to adapt M2?
I don't know the address configuration for the M2. So, I could not modify it myself.

niceboygithub, 2022-07-24:
These two bytes are "\n\n" and M2 shall returns "<RealTek>".
If there is no such string, the download won't continue.

Can you have a test?
After Aqaragateway.exe change the baudrate to 230400, close Aqaragateway.exe then use putty to open COM port with 230400 baudrate.
Check if any return characters return after press "enter" key.

joelliang, 2022-07-24:
Nop, no return character after press "enter" key.

And I monitoring the LED indicator which labeled TXD and RXD on my USB to TTL board.
   TXD: PC is sending data to M2
   RXD: M2 is sending data to PC

When "Downloading the flasher", both TXD and RXD keep flashing.
After the progress bar become 100%, the RXD flash three times.

Then I close the aqaragateway.exe, and use Putty to open the COM port with 230400 baudrate.
Every time I press "enter" key, the TXD flash once, but the RXD not flashing.
And M2 not response to all other keys on my keyboard.

niceboygithub, 2022-07-24:
Can you provide the normal boot logs?

M2 with model number ZHWG17LM may use signed firmware.
If so, flasher can not run on such CPU of M2.

joelliang, 2022-07-24:
[m2_zhwg17lm_normal_boot.log](https://github.com/niceboygithub/AqaraM1SM2fw/files/9175468/m2_zhwg17lm_normal_boot.log)

niceboygithub, 2022-07-24:
They did not change the CPU, but change SPI NAND flash.
The flasher need to be updated to add the support of new SPI NAND flash.

joelliang, 2022-07-24:
> They did not change the CPU, but change SPI NAND flash. The flasher need to be updated to add the support of new SPI NAND flash.

Glad to hear we may able use the modified firmware on the new M2.

niceboygithub, 2022-07-25:
I updated the flasher to support new SPI NAND flash.
But I do not have ZHWG17LM to test.
Can you help to test it?
```
https://github.com/niceboygithub/AqaraM1SM2fw/blob/test_new_flaser/tools/aqaragateway.exe
```

yuyamin, 2022-07-25:
> I updated the flasher to support new SPI NAND flash. But I do not have ZHWG17LM to test. Can you help to test it?
> 
> ```
> https://github.com/niceboygithub/AqaraM1SM2fw/blob/test_new_flaser/tools/aqaragateway.exe
> ```
> 
> please fix UART port to 230400 to get boot logs of flasher.

i've got 5-6 m2 gateways(17LM). Shall i try this flash tool version?

joelliang, 2022-07-26:
I could flash cfw by using the new flasher.
```
Generating padded firmware, please wait...!
Please power up gateway!
If your gateway is powered up, disconnect usb cable and reconnect it.
Downloading the flasher.
Now transmitting C:\Users\joell\AppData\Local\Temp\rootfs_3.4.4_0008.0618_modified.bin_raw
Transmit Done! Please wait for programming to flash.
Programming C:\Users\joell\AppData\Local\Temp\rootfs_3.4.4_0008.0618_modified.bin_raw Done!
```

But I got `sum:0x000` and both kernel are invalid
```
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:0, curr:0
rootfs: newest:0, curr:0
kernel[0]: sum:0x0000, size:0, fail:0
      [1]: sum:0x0000, size:0, fail:0
rootfs[0]: sum:0x0000, size:0, fail:0
      [1]: sum:0x0000, size:0, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode

nflashwrite, uiSectorCount=0 uiStartLen=55 uiEndLen=0 need_retry=0
Writing.Info: save boot_info
Info: kernel 0 is invalid
Info: kernel 1 is invalid
Warn: all kernels are invalid !
```
[cfw_boot.txt](https://github.com/niceboygithub/AqaraM1SM2fw/files/9185959/cfw_boot.txt)

joelliang, 2022-07-26:
Maybe the original FW of 17LM is different than the 12LM one.
We may need to dump and modify the 17LM FW?

joelliang, 2022-07-26:
> i've got 5-6 m2 gateways(17LM). Shall i try this flash tool version?

I don't recommend to do that, please wait for my confirmation. Or, it may brick you M2s.

joelliang, 2022-07-26:
I think it could be they change the way to calculate the sum?
Or they change the boot_info structure?

I happened to backup the original boot_info.
And keep my original FW(3.4.4_0008.0618) in rootfs_1 slot unchanged. (I flash the cfw to rootfs_0)

So, I could unbrick my M2 by flashing the original boot_info.
Here are the boot_info I backup. Please have a look.
[boot_info.zip](https://github.com/niceboygithub/AqaraM1SM2fw/files/9186185/boot_info.zip)

niceboygithub, 2022-07-26:
great that you did backup the boot_info.
You can use these commands to write back original boot_info 
```
eb 0xa0a00000 7C 91 00 00 AB 77 01 01 01 01 00 22 14 04 80 F2
eb 0xa0a00010 00 00 22 14 04 81 53 00 00 90 A0 04 DC 98 00 00
eb 0xa0a00020 94 F0 04 C8 0F 00 00 00 01 31 2E 30 2E 30 5F 30
eb 0xa0a00030 30 30 31 00 00 00 00
NANDW 0xa0000 0xa0a00000 55
```

joelliang, 2022-07-26:
I could flash back the original boot_info by using the new aqaragateway.exe
But, how can I use the modified FW?

niceboygithub, 2022-07-26:
Still checking why checksum is wrong.
If you want to unbrick your M2, just keep kernel_1 and rootfs_1 untouched. Then you can write back origin boot_info to unbrick it.

joelliang, 2022-07-26:
Previously, I'm not flashing the kernel. So the kernel and rootfs may not match on slot 0.
And then, I do the following steps:
1. flash back the original boot_info
2. flash the original 3.4.4 kernel to linux_0
3. backup the boot_info
4. flash the modified 3.3.3 rootfs to rootfs_0
5. backup the boot_info
6. boot the M2. Still no luck, check sum error.
7. backup the boot_info
[flashing_3.4.4_linux_and_rootfs.zip](https://github.com/niceboygithub/AqaraM1SM2fw/files/9186328/flashing_3.4.4_linux_and_rootfs.zip)

niceboygithub, 2022-07-26:
@joelliang 
Please use these commands to write boot_info
```
eb 0xa0a00000 7c 91 00 00 7d d1 01 01 01 00 00 22 14 04 80 f2
eb 0xa0a00010 00 00 22 14 04 81 53 00 00 94 d8 04 d2 3b 00 00
eb 0xa0a00020 94 f0 04 c8 0f 00 00 00 01 31 2e 30 2e 30 5f 30
eb 0xa0a00030 30 30 31 00 00 00 00
NANDW 0xa0000 0xa0a00000 55
```

joelliang, 2022-07-26:
Here is the first boot log just after flashing both kernel and rootfs to slot 0.
It saying `Warn: boot_info.sum error`
```
---Realtek RTL8197F boot code at 2021.07.15-15:23+0800 v3.4T-pre2.1 (993MHz)
Info: Load boot_info success!
Warn: boot_info.sum error
Info: reset boot_info.
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:0, curr:0
rootfs: newest:0, curr:0
kernel[0]: sum:0x0000, size:0, fail:0
      [1]: sum:0x0000, size:0, fail:0
rootfs[0]: sum:0x0000, size:0, fail:0
      [1]: sum:0x0000, size:0, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
```
[boot_info_sum_error.txt](https://github.com/niceboygithub/AqaraM1SM2fw/files/9186420/boot_info_sum_error.txt)

niceboygithub, 2022-07-26:
If you already flash kernel to 3.4.4 at kernel_0.
please use these commands.
```
eb 0xa0a00000 7c 91 00 00 7e 70 01 01 00 00 00 22 14 04 81 53
eb 0xa0a00010 00 00 22 14 04 81 53 00 00 94 d8 04 d2 3b 00 00
eb 0xa0a00020 94 f0 04 c8 0f 00 00 00 01 31 2e 30 2e 30 5f 30
eb 0xa0a00030 30 30 31 00 00 00 00
NANDW 0xa0000 0xa0a00000 55
```

joelliang, 2022-07-26:
> @joelliang Please use these commands to write boot_info to use cfw.
> 
> ```
> eb 0xa0a00000 7c 91 00 00 7d d1 01 01 01 00 00 22 14 04 80 f2
> eb 0xa0a00010 00 00 22 14 04 81 53 00 00 94 d8 04 d2 3b 00 00
> eb 0xa0a00020 94 f0 04 c8 0f 00 00 00 01 31 2e 30 2e 30 5f 30
> eb 0xa0a00030 30 30 31 00 00 00 00
> NANDW 0xa0000 0xa0a00000 55
> ```

These commands worked for me. And I got "Aqara-Hub-M2-B1A3 login:" at the end

joelliang, 2022-07-26:
But why curr is 1? And not 0?
```
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:1, curr:1
rootfs: newest:0, curr:1
kernel[0]: sum:0x80f2, size:2233348, fail:0
      [1]: sum:0x8153, size:2233348, fail:0
rootfs[0]: sum:0xd23b, size:9754628, fail:0
      [1]: sum:0xc80f, size:9760772, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
```

joelliang, 2022-07-26:
> If you already flash kernel to 3.4.4 at kernel_0. please use these commands.
>
> ```
> eb 0xa0a00000 7c 91 00 00 7e 70 01 01 00 00 00 22 14 04 81 53
> eb 0xa0a00010 00 00 22 14 04 81 53 00 00 94 d8 04 d2 [...]

I'm not using these commands for now. Because I could login the linux by use "admin" as username, and empty password.

niceboygithub, 2022-07-26:
I think that I already found the bug, I will upload the new flasher later.

joelliang, 2022-07-26:
> I think that I already found the bug, I will upload the new flasher later.

Glad to hear that.

BTW: it become "curr: 0" for the next boot
```
kernel: newest:1, curr:1
rootfs: newest:0, curr:0
```

joelliang, 2022-07-26:
Is there other utils or scripts needed to update as well? e.g. [fw_update](https://github.com/niceboygithub/AqaraM1SM2fw/blob/main/binutils/fw_update)

niceboygithub, 2022-07-26:
> Is there other utils or scripts needed to update as well? e.g. [fw_update](https://github.com/niceboygithub/AqaraM1SM2fw/blob/main/binutils/fw_update)

I did not get your point. Can you rephase?

KFlab, 2022-07-26:
I still  not use cable. 
and Test version not work for me. 
I really afraid for broken some thing, 

Still stop on Downloading the flasher
Or shoud i wait for more longer?

[ReceivedTofile-COM3-2022_7_26_12-15-32.log](https://github.com/niceboygithub/AqaraM1SM2fw/files/9186782/ReceivedTofile-COM3-2022_7_26_12-15-32.log)
I notise that my kernal version defferent with @joelliang 
Is that important?

Sorry that I'm too busy to  have no time to do test these days.

joelliang, 2022-07-26:
> > Is there other utils or scripts needed to update as well? e.g. [fw_update](https://github.com/niceboygithub/AqaraM1SM2fw/blob/main/binutils/fw_update)
> 
> I did not get your point. Can you rephase?

We have updated the flasher in aqaragateway.exe because the new model M2 (ZHWG17LM) using a new SPI flash.

Just a reminder.
Do we need to update other utils in your repository to address the hardware change?
e.g. fw_update and m2_update.sh

Or they could work without any change for the new M2?

niceboygithub, 2022-07-26:
What does cfw do?
1. enable telnet
2. change mosquito (mqtt) to public
3. change password to empty.

With these modifications, you can use HA aqaragateway integration.  It does not need to update other utils in this repo.

joelliang, 2022-07-26:
> I still use fixture. and Test version not work for me. I really afraid for broken some thing,
> 
> Still stop on Downloading the flasher Or shoud i wait for more longer?
> 
> [ReceivedTofile-COM3-2022_7_26_12-15-32.log](https://github.com/niceboygithub/AqaraM1SM2fw/files/9186782/ReceivedTofile-COM3-2022_7_26_12-15-32.log) I notise that my kernal version defferent with @joelliang Is that important?
> 
> Sorry that I'm too busy to have no time to do test these days. I'll try welding wires when I have free time later.

The test version not ready yet. It still has a boot_info check sum bug needed to be fixed.
Please stop trying if you don't want to brick your M2.

Please wait for the next update from @niceboygithub .

joelliang, 2022-07-26:
> What does cfw do?
> 
> 1. enable telnet
> 2. change mosquito (mqtt) to public
> 3. change password to empty.
> 
> With these modifications, you can use HA aqaragateway integration. It does not need to update other utils in this repo.

OK, thanks to the confirmation.
So, if a new version of CFW coming in the future.
I could just use the m2_update.sh to upgrade to the latest CFW.

niceboygithub, 2022-07-26:
> OK, thanks to the confirmation. So, if a new version of CFW coming in the future. I could just use the m2_update.sh to upgrade to the latest CFW.

yes

KFlab, 2022-07-26:
I welded wires from PCB and try again just now.
 both BACKUP and FLASH are not working 
with stopped on "Downloading the flasher"
and UART log seems normal

niceboygithub, 2022-07-26:
Please have a test with fix

```
https://github.com/niceboygithub/AqaraM1SM2fw/blob/new_flasher_with_fixed_boot_info/tools/aqaragateway.exe
```

yuyamin, 2022-07-26:
> Please have a test with fix
> 
> ```
> https://github.com/niceboygithub/AqaraM1SM2fw/blob/new_flasher_with_fixed_boot_info/tools/aqaragateway.exe
> ```

i couldn't get the log, However, after flashing the latest modified fw([3.4.4_0008.0618], the tool said "programming done'. Then the gateway's led-indicator is still white. I restart the gateway, the indicator: yellow-white. I guess it didn't work. May still several bugs.

joelliang, 2022-07-26:
> Please have a test with fix
> 
> ```
> https://github.com/niceboygithub/AqaraM1SM2fw/blob/new_flasher_with_fixed_boot_info/tools/aqaragateway.exe
> ```

Still `boot_info.sum error`
```
Warn: boot_info.sum error
Info: reset boot_info.
```
[boot_info_and_logs.zip](https://github.com/niceboygithub/AqaraM1SM2fw/files/9190987/boot_info_and_logs.zip)

yuyamin, 2022-07-26:
> Still `boot_info.sum error`
>
> ```
> Warn: boot_info.sum error
> Info: reset boot_info.
> ```
>
> [boot_info_and_logs.zip](https://github.com/niceboygithub/AqaraM1SM2fw/files/9190987/boot_info_and_logs.zip)

Although it can't start，after flashing the boot_info which u backuped, my gateway start successfully. I tested it with my homeassistant, it worked well.

niceboygithub, 2022-07-27:
Please help to have a test with this one.
```
https://github.com/niceboygithub/AqaraM1SM2fw/blob/new_flasher_fixed_again_with_debug/tools/aqaragateway.exe
```

Also have the logs of flasher. Fix the baudrate of UART in AccessPort  to 230400.

joelliang, 2022-07-27:
It's exactly the same as https://github.com/niceboygithub/AqaraM1SM2fw/issues/45#issuecomment-1195702656

niceboygithub, 2022-07-27:
> It's exactly the same as [#45 (comment)](https://github.com/niceboygithub/AqaraM1SM2fw/issues/45#issuecomment-1195609863)

Did you log it?
Not boot log. I need the logs while flashing. Otherwise it is hard to debug.

joelliang, 2022-07-27:
```
<RealTek>boot_ctrl set_bootinfo 64 0xfff6
Unknown command or error format: [set_bootinfo]
```
[flashing_logs.txt](https://github.com/niceboygithub/AqaraM1SM2fw/files/9195923/flashing_logs.txt)

niceboygithub, 2022-07-27:
> ```
> <RealTek>boot_ctrl set_bootinfo 64 0xfff6
> Unknown command or error format: [set_bootinfo]
> ```
> 
> [flashing_logs.txt](https://github.com/niceboygithub/AqaraM1SM2fw/files/9195923/flashing_logs.txt)

This log is that you try to flash boot_info by AqaraGateway. This function is not supported.
I need the logs that flashing rootfs.

joelliang, 2022-07-27:
Here it's
[flashing_rootfs_logs.txt](https://github.com/niceboygithub/AqaraM1SM2fw/files/9196206/flashing_rootfs_logs.txt)

niceboygithub, 2022-07-27:
There is no problem on checksum.
Did you flash other kernel, or boot_info after flashed rootfs?

joelliang, 2022-07-27:
The rootfs I flashed into rootfs_0 is the same of the already in the SPI flash.
So, the size and check sum should be the same.
```
<RealTek>boot_ctrl set_rootfs0 9754628 0xd23b
Setting rootfs 0 size: 9754628 -> 9754628
Setting rootfs 0 checksum: 3bd23b -> 0xd23b
```
But you changed the checksum from `3bd23b` to `0xd23b`.
Are you using int16 to store a int32 data?

niceboygithub, 2022-07-27:
> The rootfs I flashed into rootfs_0 is the same as the one already in the SPI flash. So, the size and check sum should be the same.
> 
> ```
> <RealTek>boot_ctrl set_rootfs0 9754628 0xd23b
> Setting rootfs 0 size: 9754628 -> 9754628
> Setting rootfs 0 checksum: 3bd23b -> 0xd23b
> ```
> 
> But you changed the checksum from `3bd23b` to `0xd23b`. Are you using int16 to store a int32 data?

I think that it is just display problem
I saw the problem

While boot_ctrl set_rootfs0 9754628 0xd23b
The boot_info is
```
7c ffffff91 00 00 7d ffffffd2 01 00 01 00 00 22 14 04 ffffff80 fffffff2 00 00 22 14 04 ffffff81 53 00 00 ffffff94 ffffffd8 04 ffffffd2 3b 00 00 ffffff94 fffffff0 04 ffffffc8 0f 00 00 00 01 31 2e 30 2e 30 5f 30 30 30 31 00 00 00 00
```
The checksum is '7d d2' (ignore ffffff)

Next command is 'Setting newest rootfs: 0 -> 0'
The boot_info is
```
7c ffffff91 00 00 7d 32 01 00 01 00 00 22 14 04 ffffff80 fffffff2 00 00 22 14 04 ffffff81 53 00 00 ffffff94 ffffffd8 04 ffffffd2 3b 00 00 ffffff94 fffffff0 04 ffffffc8 0f 00 00 00 01 31 2e 30 2e 30 5f 30 30 30 31 00 00 00 00
```
The checksum is '7d 32' and it is wrong.

joelliang, 2022-07-27:
> There is no problem on checksum. Did you flash other kernel, or boot_info after flashed rootfs?

I did flash 3.4.4 kernel to linux_0, and I'm not using the commands you told me at https://github.com/niceboygithub/AqaraM1SM2fw/issues/45#issuecomment-1194983567

Is that the problem?

Should I flash both kernel and rootfs again?

[25 more replies]

closed 2022-08-02 by niceboygithub