niceboygithub/AqaraM1SM2fw #49: M2 HM2-G01 No boot

BarakaAka1Only, 2022-08-15:
So i been trying to flash my M2 to enable telnet and i had no success. 

The issue is when i flashed the modded firmware on rootfs_0 the flash apparently was successful with Done! 

So i waited a few minutes just in-case nothing was else needed for the aqaragateway.exe to do and i unplugged my unit and replugged it in to try to test if telnet works now..

However I can't get the unit to start up because in the UART logs it states that both rootfs 0 and 1 doesn't exist and i went and looked here at all the open /closed issues and found some commands to run so that the cfw can boot.

No i didn't do anything with the kernel nor the linux partitions only i touched rootfs_0.

Apparently one user stated to flash it on 1 as it helped him but i don't want to touch it if it makes the issue even more a problem

```
uart ok
strap:0x412b8ae2
enable spi-nand
ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 MHz)

Found recognized ID, rdid=0x00efaa21
init IP fail(0xffffffff)
init ddr ok
img sig ok
chksum ok
load img ok
s-boot
sec sig ok
decrypt img
jump 0xa0000000

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400

[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512

[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!

check v2r bbt table:0 OK

[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704

[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!

check bbt table:0 OK

[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
Realtek Crypto Engine v0.1
=>CPU Wake-up interrupt happen! GISR=09000084 
 
---Realtek RTL8197F boot code at 2020.07.14-20:40+0800 v3.4T-pre2.2 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:1, curr:1
rootfs: newest:0, curr:1
kernel[0]: sum:0x0000, size:2359296, fail:3
      [1]: sum:0x8153, size:2233348, fail:3
rootfs[0]: sum:0x0000, size:9830400, fail:0
      [1]: sum:0xc80f, size:9760772, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: kernel 1 is invalid
Info: kernel 0 is invalid
Warn: all kernels are invalid !

---Ethernet init Okay!
```

So in UART i ran 

```
eb 0xa0a00000 7c 91 00 00 f4 b7 00 00 00 00 00 22 14 44 80 bc
eb 0xa0a00010 00 00 22 14 44 80 bc 00 00 90 20 44 fb a6 00 00
eb 0xa0a00020 90 20 44 fb a6 00 00 00 01 31 2e 30 2e 32 2e 30
eb 0xa0a00030 30 35 00 00 00 00 00
NANDW 0xa0000 0xa0a00000 55
```

Didn't work as i get now 

```
uart ok
strap:0x412b8ae2
enable spi-nand
ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 MHz)

Found recognized ID, rdid=0x00efaa21
init IP fail(0xffffffff)
init ddr ok
img sig ok
chksum ok
load img ok
s-boot
sec sig ok
decrypt img
jump 0xa0000000

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400

[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512

[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!

check v2r bbt table:0 OK

[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704

[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!

check bbt table:0 OK

[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
Realtek Crypto Engine v0.1
=>CPU Wake-up interrupt happen! GISR=09000084 
 
---Realtek RTL8197F boot code at 2020.07.14-20:40+0800 v3.4T-pre2.2 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:0, curr:0
rootfs: newest:0, curr:0
kernel[0]: sum:0x80bc, size:2233412, fail:0
      [1]: sum:0x80bc, size:2233412, fail:0
rootfs[0]: sum:0xfba6, size:9445444, fail:0
      [1]: sum:0xfba6, size:9445444, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode

nflashwrite, uiSectorCount=0 uiStartLen=55 uiEndLen=0 need_retry=0
Writing.Info: save boot_info
Info: loading kernel 0 ... size 2233412
Info: checking kernel 0 ... 
Info: loading kernel 1 ... size 2233412
Info: checking kernel 1 ... 
Success!
Info: loading rootfs 0 ...  Done
Info: checking rootfs 0 ... Info: checking rootfs 0 ... 
Fail
Info: loading rootfs 1 ...  Done
Info: checking rootfs 1 ... Info: checking rootfs 1 ... 
Fail
Warn: no rootfs available.

nflashwrite, uiSectorCount=0 uiStartLen=55 uiEndLen=0 need_retry=0
Writing.Info: save boot_info

---Ethernet init Okay!
<RealTek>
```

Tried

```
eb 0xa0a00000 7c 91 00 00 e5 a8 00 00 00 00 00 22 14 44 80 bc
eb 0xa0a00010 00 00 22 14 44 80 bc 00 00 90 20 44 fb b5 00 00
eb 0xa0a00020 90 20 44 fb b5 00 00 00 01 31 2e 30 2e 32 2e 30
eb 0xa0a00030 30 35 00 00 00 00 00
NANDW 0xa0000 0xa0a00000 55
```
Didn't work either...

Tried with both the original firmware and the modded firmwares just to be sure and all my unit does via the LED

Goes red then after a few seconds yellow then solid white

No i didn't backup anything because i wasn't sure exactly how do as the button Backup did nothing for me.

I am using Free Serial Analyzer to view the console at 38400 with 8 bits and i am using a CP210x device

```
eb 0xa0a00000 7c 91 00 00 7d d1 01 01 01 00 00 22 14 04 80 f2
eb 0xa0a00010 00 00 22 14 04 81 53 00 00 94 d8 04 d2 3b 00 00
eb 0xa0a00020 94 f0 04 c8 0f 00 00 00 01 31 2e 30 2e 30 5f 30
eb 0xa0a00030 30 30 31 00 00 00 00
NANDW 0xa0000 0xa0a00000 55
```

States 
```
---Realtek RTL8197F boot code at 2020.07.14-20:40+0800 v3.4T-pre2.2 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:1, curr:1
rootfs: newest:0, curr:1
kernel[0]: sum:0x80f2, size:2233348, fail:0
      [1]: sum:0x8153, size:2233348, fail:0
rootfs[0]: sum:0xd23b, size:9754628, fail:0
      [1]: sum:0xc80f, size:9760772, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: loading kernel 1 ... size 2233348
Info: checking kernel 1 ... 
Info: loading kernel 0 ... size 2233348
Info: checking kernel 0 ... 
Warn: all kernels are invalid !
```

So i am at a lost of what and why this is happing and before i started all of this yes my unit was on the the latest firmware via 3.4.4_0008.0618 as shown:

[image]

Any help with this would be greatly appreciated

niceboygithub, 2022-08-16:
As I said in README, the first notice.

**Notice: These modified firmwares do NOT support EU version.** 

You can flash EU rootfs (which was signed) to save your bricked M2.

CFU is not signed fw.

BarakaAka1Only, 2022-08-16:
> As I said in README, the first notice.
> 
> 
> 
> **Notice: These modified firmwares do NOT support EU version.** 
> 
> 
> 
> You can flash EU rootfs (which was signed) to save your bricked M2.
> 
> 
> 
> CFU is not signed fw.

I am aware of that but I am don't have a EU version I purchased this via Amazon and I am here in the US 

So then I'll go and flash the EU now I been doing the normal non EU ...

niceboygithub, 2022-08-16:
As I know, HM2-G01 is EU version and it use signed fw.

In my local market, there are CN and EU versions of M2. So it needs to check model before flash cfw.

BTW, there will be another EU model of M2 and it does not use signed fw.
I will update the README to describe it.

BarakaAka1Only, 2022-08-16:
> As I know, HM2-G01 is EU version and it use signed fw.
> 
> In my local market, there are CN and EU versions of M2. So it needs to check model before flash cfw.
> 
> BTW, there will be another EU model of M2 and it does not use signed fw. I will update the README to clarify it.

Had no idea. I apologize as i wasn't trying to waste your time and i know it sucks to constantly repeat yourself. 
I can indeed confirm i have the EU variant flashed back to 3.3.0_0021.0618 no issues thank you for saving my device. Gonna have to see how to manually set for persist.app.tty_enable

niceboygithub, 2022-08-16:
> Had no idea. I apologize as i wasn't trying to waste your time and i know it sucks to constantly repeat yourself. I can indeed confirm i have the EU variant flashed back to 3.3.0_0021.0618 no issues thank you for saving my device. 

It is ok to discuss. I am first time to know that ppl can buy HM2-G01 in US market.

> Gonna have to see how to manually set for persist.app.tty_enable

If you found the method, please let me know.

rezmus, 2022-08-19:
eu = global (also us, ru, sg, au). not cn ;)

r3knit, 2022-08-28:
@niceboygithub 
Hi!
I have something similar to this topic

My friend gave me a bricked (hope not) M2 that he tried to flash.. Of course, it was EU version.
I dont know, what he have done with it but now it has this output on serial startup

```
uart ok
strap:0x412b8ae2
enable spi-nand
ROM ver:v1.21, sig:866c151, time:2016.11.04-11:26+0800, CPU(400 MHz), DDR2(533 MHz)

Found recognized ID, rdid=0x00efaa21
init IP fail(0xffffffff)
init ddr ok
img sig ok
chksum ok
load img ok
s-boot
sec sig ok
decrypt img
jump 0xa0000000

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400
[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512
[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!
check v2r bbt table:0 OK
[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704
[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!
check bbt table:0 OK
[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
Realtek Crypto Engine v0.1
=>CPU Wake-up interrupt happen! GISR=09000084

---Realtek RTL8197F boot code at 2020.07.14-20:40+0800 v3.4T-pre2.2 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:0, curr:0
rootfs: newest:0, curr:0
kernel[0]: sum:0x80bc, size:2233412, fail:0
      [1]: sum:0x8082, size:2233412, fail:0
rootfs[0]: sum:0xfbb5, size:9445444, fail:3
      [1]: sum:0xfbb5, size:9445444, fail:3
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: loading kernel 0 ... size 2233412
Info: checking kernel 0 ...
Success!
Info: rootfs 0 is invalid
Info: rootfs 1 is invalid
Warn: no rootfs available.

---Ethernet init Okay!
<RealTek>
```

I've tried to flash with original signed fw, but no luck there. And yes, of course, there is no backups 

Could you please help me to revive it?

BarakaAka1Only, 2022-08-28:
@r3knit you have to edit the "bootloader header" after you flash the signed EU firmware with whatever UART tool you are using when you plug in the unit make sure the  console is at 38400 with 8 bits and as immediately when you power up the unit hit  the u key on the keyboard until you see <RealTek> show up the try these 

```
eb 0xa0a00000 7c 91 00 00 f4 b7 00 00 00 00 00 22 14 44 80 bc
eb 0xa0a00010 00 00 22 14 44 80 bc 00 00 90 20 44 fb a6 00 00
eb 0xa0a00020 90 20 44 fb a6 00 00 00 01 31 2e 30 2e 32 2e 30
eb 0xa0a00030 30 35 00 00 00 00 00
NANDW 0xa0000 0xa0a00000 55
```

or 

```
eb 0xa0a00000 7c 91 00 00 e5 a8 00 00 00 00 00 22 14 44 80 bc
eb 0xa0a00010 00 00 22 14 44 80 bc 00 00 90 20 44 fb b5 00 00
eb 0xa0a00020 90 20 44 fb b5 00 00 00 01 31 2e 30 2e 32 2e 30
eb 0xa0a00030 30 35 00 00 00 00 00
NANDW 0xa0000 0xa0a00000 55
```

Pick one and copy and paste one by one line after line with hitting enter at the end then restart the unit if doesn't work use the other.

i did state this above but not in detail, hope this helps you.

r3knit, 2022-08-28:
@BarakaAka1Only
Thank you so much. It helped!

BarakaAka1Only, 2022-08-29:
> @BarakaAka1Only Thank you so much. It helped!

No problem at all :)

closed 2022-08-31 by niceboygithub