niceboygithub/AqaraM1SM2fw #50: Success on EU version M2 gateway

himekifee, 2022-09-08:
Even though it's said to be incompatible with EU version of M2 gw, I still managed to flash the rootfs using the Chinese modified ver. Result as below.

[image]
[image]
[image]
[image]

BarakaAka1Only, 2022-09-10:
Interesting, well i went ahead and tried it and it failed console comes out with no root_fs. Restoring root_3_3_0_0021_0526.bin via signed works only for me. So i wonder what default firmware you were running before the flash?

Did you edit the boot_info?

himekifee, 2022-09-10:
I forgot which one but I updated to the latest in aqara home app before flashing.

BarakaAka1Only, 2022-09-10:
> I forgot which one but I updated to the latest in aqara home app before flashing.

Gotcha okay,  gonna flash back mine to root_3_3_0_0021_0526 then do an update on the app and then flash the modified fw to see what occurs.

BarakaAka1Only, 2022-09-10:
Did not work at all so i am shocked

himekifee, 2022-09-10:
Any screenshoots for me to have a look?

BarakaAka1Only, 2022-09-10:
No as it just does 
```
Info: loading rootfs 0 ...  Done
Info: checking rootfs 0 ... Info: checking rootfs 0 ... 
Fail
Info: loading rootfs 1 ...  Done
Info: checking rootfs 1 ... Info: checking rootfs 1 ... 
Fail
Warn: no rootfs available.
```

and 

```
Info: loading kernel 1 ... size 2233348
Info: checking kernel 1 ... 
Info: loading kernel 0 ... size 2233348
Info: checking kernel 0 ... 
Warn: all kernels are invalid !
```

if i edit the boot_info

himekifee, 2022-09-10:
Could you get a acreenshot of that aqaragateway.exe?

BarakaAka1Only, 2022-09-10:
[image]

 Currently i am restoring AIOT_V3.4.6_0010.0618_20220822

himekifee, 2022-09-10:
Did it success on flashing? I used a [monitor](https://freeserialanalyzer.com/) to check all com port traffics. You may want to have a look at the packets sent back.

BarakaAka1Only, 2022-09-10:
I do actually know what i am doing as far as that goes, and yes it did complete 

Example:
```
Do NOT power up your gateway.
IF it is already power up, remove power cable!

Generating padded firmware, please wait...!
Please power up gateway!
If your gateway is powered up, disconnect usb cable and reconnect it.
Downloading the flasher.
Now transmitting C:\Temp\root.bin_raw
Transmit Done! Please wait for programming to flash.
Programming C:\Temp\root.bin_raw Done!
Do NOT power up your gateway.
IF it is already power up, remove power cable!
```

Baud Rate is 38400 ... and not using that tool i am on a Mac to do all my monitoring as on my windows machine free doesn't apply with modbus

himekifee, 2022-09-10:
Maybe you can try flash the stock chinese version? To mention that after flashing the modified firmware, I used that update.sh to crossflash. It ends up with all functionalities are working and it show as a chinese model from getprop.

himekifee, 2022-09-10:
Or maybe simply just two batch of hardwarse with two different pcbs. I bought it from UK amazon 3 days ago.

BarakaAka1Only, 2022-09-10:
After many attempts flashing other firmwares both EU and CN nothing works only the signed firmware(s) works as stated many times by @niceboygithub 

I myself got mine via Amazon and i am in the USA yet mine is and looks exactly like yours so not sure whats the case in this and with the update.sh doesn't do anything for me, however i can go from 3.3.0_0021_0526 (original) to  3.4.6_0007.0618 M2 CN original and it boots up but again with the modified fw doesn't boot but it reports the version as 3.4.6_0010.0618 even tho M2 CN 3.4.6_0007.0618 (original) was flashed on rootfs_1 and on rootfs_0 its in invalid state

himekifee, 2022-09-10:
Maybe because of the bootloader. If you want I can dump whatever you want.

BarakaAka1Only, 2022-09-10:
After closely looking at yours to mine, this is how mines looks like 

[image]

and cool if you could dump it all would be great.

Nismonx, 2022-09-10:
I got mine from amazon UK several months ago, if I have a little spare time I can try the same steps as suggested. Would be interesting what happens.

BarakaAka1Only, 2022-09-11:
After looking back, @niceboygithub did state and i quote "BTW, there will be another EU model of M2 and it does not use signed fw." so to me @himekifee you may got the EU version that doesn't use signed FW which is the issue on my end, mine is signed. Well just gonna have to wait or until i am able to find a way to get telnet, would be great to get a CN version or even the one you have but aliexpress doesn't have them and amazon is a hit or miss..

himekifee, 2022-09-11:
Do you wanna maybe try flashing my bootloader? I'm still finding a way to extract that.

BarakaAka1Only, 2022-09-11:
I am open to trying anything at this point.

himekifee, 2022-09-11:
So what you got now? Do you have any access to a shell inside? Otherwise I don't really think you have a way to write to nand

BarakaAka1Only, 2022-09-11:
Only serial is what i have at this point but its better to have a dump, i do have some tools to dump the winbond chip but i haven't gotten around to doing that at all

himekifee, 2022-09-11:
I dumped some bit using `dd`. There are some errors when dumping using `nanddump` so have to do that. I think you are more experienced on this so I just give all the mtd dump(except data partition) to you to have a look.
[mtd.zip](https://github.com/niceboygithub/AqaraM1SM2fw/files/9542743/mtd.zip)

rezmus, 2022-09-14:
@himekifee can you do getprop | grep lumi.gateway? there is 2nd gen eu hub lumi.gateway.agl001 100% same as 1st gen cn. that would explain why not signed rootfs works for you.

himekifee, 2022-09-14:
```
# getprop | grep lumi.gateway
[persist.sys.model]: [lumi.gateway.agl001]
[ro.sys.model]: [lumi.gateway.iragl5]
```
Note that I flashed Chinese modified firmware.

rezmus, 2022-09-14:
it does not matter because fw does not touch device db (stored on userdata and restored from /res mtd). as you can see you have 2nd gen eu hub agl001, it has 100% same firmware as cn 1st gen cn iragl5. unfortunately it does not change anything for owners of 1st gen eu irag7 with signed rootfs.

rezmus, 2022-09-15:
@himekifee can you check production date from sticker? may be handy for some users, because last batch of irag7 was probably made months ago.

himekifee, 2022-09-15:
SKU: AG035EUB01
Production Date: 2022/05

rezmus, 2022-09-15:
1st gen m2 eu has sku AG022GLB02.

BarakaAka1Only, 2022-09-17:
I went and took a risk on Amazon and purchased a new unit and looks like its the same as @himekifee but its the US version from that the box states.

I took some photos:

[image]
[image]
[image]
[image]
[image]
[image]

Here is the first boot UART Console Log:

```
uart ok
strap pin:0x412b8ae2
enable spi-nand
ROM ver:v1.1, sig:455cc27, time:2016.01.04-18:42+0800, CPU(400 MHz), DDR2(533 MHz)
load efuse ok
init IP ok
img sig ok
chksum ok
load img ok
jump 0xbfe01550

Booting...
SPI NAND clock not enable

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400
[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512
[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!
check v2r bbt table:0 OK
[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704
[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!
check bbt table:0 OK
[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
=>CPU Wake-up interrupt happen! GISR=89000084 
 
---Realtek RTL8197F boot code at 2021.12.06-21:28+0800 v3.4T-pre2.1 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:0, curr:0
rootfs: newest:0, curr:0
kernel[0]: sum:0x81cb, size:2233348, fail:0
      [1]: sum:0x81cb, size:2233348, fail:0
rootfs[0]: sum:0xb7ea, size:9515012, fail:0
      [1]: sum:0xb7ea, size:9515012, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: loading kernel 0 ...  Done
Info: checking kernel 0 ... Success
Info: select rootfs 0
cmdline:root=/dev/mtdblock5 console=ttyS0,38400
[... 203 lines cut ...]
[   23.510000] [97F] Bonding Type 97FS, PKG1
[   23.510000] [97F] RFE type 0 PHY paratemters: DEFAULT
[   23.510000] clock 40MHz
[   23.510000] load efuse ok
[   23.510000] rom_progress: 0x200006f
[   23.510000] rom_progress: 0x400006f
[   23.580000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0] size
[   23.600000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0]
[   23.820000] [GetHwReg88XX][rtl8197Ffw]
[   23.830000] [GetHwReg88XX][rtl8197Ffw size]
[   24.030000] Virtual device wlan0 asks to queue packet!
[   24.290000] [97F] Default BB Swing=30
```

In the log you can see it states:

```
fw_manager.sh revision: 2
pre-revision: 2

hostname: Aqara-Hub-M2-003B
type: lumi.gateway.agl001, model: AH_M2_BLE
```

I am going to flash it and see what happens, i just barely got my hands on it and gained UART..

BarakaAka1Only, 2022-09-17:
Flashed rootfs_3.3.5_0012.0611_modified.bin just to keep the "version" the same 

[image]

And it worked 

```
uart ok
strap pin:0x412b8ae2
enable spi-nand
ROM ver:v1.1, sig:455cc27, time:2016.01.04-18:42+0800, CPU(400 MHz), DDR2(533 MHz)
load efuse ok
init IP ok
img sig ok
chksum ok
load img ok
jump 0xbfe01550

Booting...
SPI NAND clock not enable

SPI Nand ID=00efaa21
SPI Nand die chipsize=0x08000000 byte
SPI Nand dienum=1,
SPI Nand blocksize=0x00020000 byte,
SPI Nand pagesize=0x00000800 byte,
SPI Nand oobsize=0x00000040 byte,
[rtkn_scan_bbt, line 1812], RBA=51, this->RBA_PERCENT = 5,block_v2r_num=1024
[rtkn_scan_bbt, line 1822] block_v2r_num 00000400
[rtk_scan_v2r_bbt]:678,RBA=00000033,2=00000400,
[rtk_scan_v2r_bbt]:684,block_v2r_num=000003cd
INFO: Stored BBT in Die 0: block=8 , block_status_p1=0x000000bb
load bbt v2r table:0 page:512
[rtk_scan_v2r_bbt] have created v2r bbt table:0 on block 8, just loads it !!
check v2r bbt table:0 OK
[rtk_nand_scan_bbt, line 393] mem_page_num=1 bbt_page 704
INFO: Stored BBT in Die 0: block=11 , block_status_p1=0x000000bb
load bbt table:0 page:704
[rtk_nand_scan_bbt] have created bbt table:0 on block 11, just loads it !!
check bbt table:0 OK
[dump_BBT] Nand BBT Content
Congratulation!! No BBs in this Nand.
=>CPU Wake-up interrupt happen! GISR=09000084 
 
---Realtek RTL8197F boot code at 2021.12.06-21:28+0800 v3.4T-pre2.1 (993MHz)
Info: Load boot_info success!
== RTL8197 Aqara Gateway bootloader ==
boot_info: ver:0
kernel: newest:0, curr:0
rootfs: newest:0, curr:0
kernel[0]: sum:0x81cb, size:2233348, fail:0
      [1]: sum:0x81cb, size:2233348, fail:0
rootfs[0]: sum:0xa7a4, size:9537540, fail:0
      [1]: sum:0xa7a4, size:9537540, fail:0
root_sum_check: off
watchdog_time: 0
boot_version: 1.0.0_0001
boot_magic: 0000917c
priv mode
Info: loading kernel 0 ...  Done
Info: checking kernel 0 ... Success
Info: select rootfs 0
cmdline:root=/dev/mtdblock5 console=ttyS0,38400
Info: booting...
Jump to image start=0x80a00000...
decompressing kernel:
Uncompressing Linux... done, booting the kernel.
done decompressing kernel.
start address: 0x804e4f10
[    0.000000] Linux version 3.10.90 (huangpingjie-e0051@compilex64-ipg) (gcc version 4.9.4 20151028 (prerelease) (Realtek MSDK-4.9.4p1 Build 2648) ) #18 Tue Dec 28 14:46:15 CST 2021
[... 232 lines cut ...]
Jan  1 00:00:17 ha_agent[955]: handle_register: method [basis.network] is registered for role(4) (ha_msg.cpp:115)
Jan  1 00:00:17 ha_agent[955]: handle_register: method [basis.argb] is registered for role(4) (ha_msg.cpp:115)
Jan  1 00:00:17 ha_agent[955]: handle_register: method [basis.state] is registered for role(4) (ha_msg.cpp:115)
Jan  1 00:00:17 ha_agent[955]: handle_register: method [auto.action] is registered for role(4) (ha_msg.cpp:115)
Jan  1 00:00:17 ha_agent[955]: handle_register: method [auto.control] is registered for role(4) (ha_msg.cpp:115)
[   18.160000] WlanSupportAbility = 0x3
[   18.170000] [ODM_software_init] 
[   18.170000] [97F] Bonding Type 97FS, PKG1
[   18.170000] [97F] RFE type 0 PHY paratemters: DEFAULT
[   18.170000] clock 40MHz
[   18.170000] load efuse ok
[   18.170000] rom_progress: 0x200006f
[   18.170000] rom_progress: 0x400006f
[   18.240000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0] size
[   18.260000] [GetHwReg88XX][PHY_REG_PG_8197Fmp_Type0]
[   18.470000] [GetHwReg88XX][rtl8197Ffw]
[   18.490000] [GetHwReg88XX][rtl8197Ffw size]
[   18.930000] [97F] Default BB Swing=30
```

Shell Access:

[image]

So i can confirm Units with Production Date: 2022/05  are unsigned @niceboygithub

BarakaAka1Only, 2022-09-17:
Updated using m2_update.sh script which i did have to modify for it to work 

curl is required 

```
mkdir /data/bin
wget -O /data/bin/curl "http://master.dl.sourceforge.net/project/mgl03/bin/curl?viasf=1"; chmod +x /data/bin/curl
cp /data/bin/curl /tmp/curl

```
then use the modified script [here](https://drive.google.com/file/d/1-uVvxb2kMSpWHtbyfxCF3yhoEsH9XU5-/view?usp=sharing)  

and make sure to run ```chmod +x m2_update.sh ```

rezmus, 2022-09-17:
also when sku starts with AG035 is new agl001 version.

artoodetoo3301, 2022-10-07:
Sorry, I desperately need help on the steps as it is quite unclear.

I downloaded the Aqara Gateway Utils, using uart tiny board flashing the M2 hub.
I chose rootfs_1, xmodem, rootfs_3.4.6_0007.0618_modified.bin, following the steps until Programming C:\~~~~~~~~\rootfs_3.4.6_0007.0618_modified.bin_raw Done!

After this do I need to do anything else?

Then I tried disconnect the flashing tiny board, unplug and plugin the M2 Hub again, the LED started as Red, then it is yellow forever, no blinking, no flashing, stay always on. Did I brick my device? The Aqara app cannot search the Hub anymore.

Did I miss something, how could I fix it?

Thanks.
[image]

m32w, 2022-12-13:
Hello I am planing to flash UE version (without UE\UK mark), can you let me know which PINs did you used ?
[image]
[image]

Nismonx, 2022-12-13:
@m32w  I don't think you have the right hub.

closed 2024-01-25 by niceboygithub

JCalvi, 2024-09-23:
@himekifee or @BarakaAka1Only, can you join non Chinese servers after flashing the hub?

JCalvi, 2024-09-26:
I have flashed my HM2-G01 with Aqaragateway and updated via m2_update.sh to 4.18 and have joined Australia server and all working through Home Assistant and Homekit as well.