niceboygithub/AqaraM1SM2fw #57: Aqara P3 UART TTL and power pins

665465, 2022-11-13:
I just missed flash a wrong firmware into Aqara P3. Now the device doesn't boot anymore. I opened the case but the UART pins are n't labeled. 
Could someone share the UART and power pins position ? I want to try flash firmware by xmodem into P3.
Thanks.
[image]

Nismonx, 2022-11-13:
I don't have one so can't help you much. Just though of mentioning if you lookup the MCU schematics and with the help of a multimeter you should be able to narrow it down. Or just wait for someone to come and help ¯\_(ツ)_/¯

665465, 2022-11-21:
> I don't have one so can't help you much. Just though of mentioning if you lookup the MCU schematics and with the help of a multimeter you should be able to narrow it down. Or just wait for someone to come and help ¯_(ツ)_/¯

Thanks.
I've found the GND/Power and Tx pin. Not sure about the Rx because the device stuck in boot loop and is not responding to key input.
[image]

[image]

And it seems there is no option in aqaragateway tool for Aqara P3 gateway. Is there any other way I can flash the FW into the gateway?
[image]

niceboygithub, 2022-12-19:
Can you provide the log that can not boot up?

665465, 2022-12-19:
Please see the attachment for the full log. I flashed a Xiaomi gateway 3 firmware into P3 by mistake. Seems there is even no way to get into uboot?
[full_log.txt](https://github.com/niceboygithub/AqaraM1SM2fw/files/10258443/full_log.txt)

niceboygithub, 2022-12-31:
You can flash P3 firmware by choose M1S model.

closed 2023-08-18 by niceboygithub