niceboygithub/AqaraM1SM2fw #58: H1 gateway modified firmware latest version

miniknife88, 2022-12-17:
Hopefully the latest version of the modified firmware can be provided，please tell me how to extract the original firmware

niceboygithub, 2022-12-19:
Do you flash the old modified firmware?

miniknife88, 2022-12-19:
yes，what do i need to do？currently using version 3.4.0_0004.0617，latest version 3.5.0_0013.0736

niceboygithub, 2022-12-19:
My way, (not best way)

1.  telnet to H1
```
setprop persist.app.debug_log true;
getprop persist.app.debug_log
reboot
```
check by getprop to make it is tru
2. telnet to H1 again
```
tail /tmp/*.txt | grep http
```
3. go to aqara home to do firmware upgrade
4. when you see the url in step, ctrl-break the tail command and reboot immediately to prevent it update to last firmware.

miniknife88, 2022-12-19:
ctrl-break the tail command and reboot，that's it?

wcl272560, 2022-12-31:
The URL has been obtained according to your steps

https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.sacn01/20221207180313_lumi.gateway.sacn01_AIOT_V3.5.0_0013.0736_20221205_c25285.bin

what to do next?

niceboygithub, 2022-12-31:
I already uploaded the firmware.

closed 2023-02-28 by niceboygithub