niceboygithub/AqaraM1SM2fw #61: Any modified firmware for M1S 2022 CN?

ashottonoyan, 2023-01-22:


SikaiGGMU, 2023-01-31:
it seems like M1S 2022 CN changed its soc, and the instruction architecture switches from mips32 to arm v7

niceboygithub, 2023-02-28:
No plan to create modified firmware for M1S 2022 CN.
M1S 2022 CN was supported in AlexxIT's [openmiio_agent](https://github.com/AlexxIT/openmiio_agent)
The openmiio_agent can run in M1S 2022 CN.

SikaiGGMU, 2023-05-12:
Will the uart tx rx layout of M1S2022CN be published in the near future ?

SikaiGGMU, 2023-05-16:
I accidently erased my active rootfs block, which turned my device into a 'brick'. Probably, only uart can save my device.

SikaiGGMU, 2023-05-16:
I accidently erased my active rootfs block, which turned my device into a 'brick'. Probably, only uart can rescue my device.

SikaiGGMU, 2023-06-05:
Hi, will it be published?

niceboygithub, 2023-06-05:
Publish what?

SikaiGGMU, 2023-06-06:
uart tx rx layout of M1S2022CN

closed 2023-08-14 by niceboygithub