niceboygithub/AqaraM1SM2fw #63: [New firmware support]: AIOT_V3.5.2_0010.0736 for Aqara Hub H1

ashottonoyan, 2023-02-28:
https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.sacn01/20230207112107_lumi.gateway.sacn01_AIOT_V3.5.2_0010.0736_20230206_d2135c.bin

ashottonoyan, 2023-03-01:
Thanks for the upload. An error in the script causes the model not to be determined.

```
➜  ~ telnet 10.100.1.34
Connected to 10.100.1.34

Entering character mode
Escape character is '^]'.

rlxlinux-D7D2 login: admin
RLX Linux version 3.0
         _           _  _
        | |         | ||_|                 
   _  _ | | _  _    | | _ ____  _   _  _  _ 
  | |/ || |\ \/ /   | || |  _ \| | | |\ \/ /
  | |_/ | |/    \   | || | | | | |_| |/    \
  |_|   |_|\_/\_/   |_||_|_| |_|\____|\_/\_/

For further information check:
http://processor.realtek.com/

BusyBox v1.22.1 (2022-12-05 20:39:29 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

# cd /tmp && wget -O /tmp/curl "http://master.dl.sourceforge.net/project/mgl03/bin/curl?viasf=1" && chmod a+x /tmp/curl
-s -k -L -o /tmp/h1_update.sh https://raw.githubusercontent.com/niceboygithub/AqaraM1SM2fw/main/modified/H1/h1_update.sh
chmod a+x /tmp/h1_update.sh && /tmp/h1_update.shConnecting to master.dl.sourceforge.net (216.105.38.12:80)
curl                 100% |*************************************************************************************************************************************************|  2271k  0:00:00 ETA
# /tmp/curl -s -k -L -o /tmp/h1_update.sh https://raw.githubusercontent.com/niceboygithub/AqaraM1SM2fw/main/modified/H1/h1_update.sh
# chmod a+x /tmp/h1_update.sh && /tmp/h1_update.sh
h1_update.sh revision: 2
type: lumi.gateway.sacn01, model: 
platform: aiot, path: /tmp, sign: 0
Unpack path : /tmp/ota_dir
Firmware path : /data/ota_dir
Available ROM size(b): 34525184
Available RAM size(b): 25460736
Error: Unknow model
Update failed, reason: getpack failed!
```

ashottonoyan, 2023-03-01:
fixed and created a pull request

niceboygithub, 2023-03-02:
thanks for your efforts.

closed 2023-03-07 by ashottonoyan