niceboygithub/AqaraM1SM2fw #83: P3 update fail

SaintWe, 2023-08-06:
```
.....
# chmod a+x /tmp/p3_update.sh && /tmp/p3_update.sh
p3_update.sh revision: 2
type: lumi.aircondition.acn05, model: AC_P3
platform: aiot, path: /tmp/fw.tar.gz, sign: 0
Unpack path : /tmp/ota_dir
Firmware path : /data/ota_dir
Available ROM size(b): 26259456
Available RAM size(b): 27131904
Update to 4.0.2_0024.0644
Get packages, please wait...
Got packages done
AalUart_Open /dev/ttyS2
recv STA!
recv STA!
recv close ACK!
recv CRC check!
ir_loacl_crc = 962b, ir_msg_crc=72da,retry...
recv STA!
Error:Unkonw flag:00000043!
ir_loacl_crc = 0, ir_msg_crc=72da,retry...
recv STA!
Error:Unkonw flag:00000043!
ir_loacl_crc = 0, ir_msg_crc=72da,retry...
recv STA!
Error:Unkonw flag:00000043!
ir_loacl_crc = 0, ir_msg_crc=72da,retry...
recv STA!
Error:Unkonw flag:00000043!
ir_loacl_crc = 0, ir_msg_crc=72da,retry...
Update failed, reason: OTA failed!
```

niceboygithub, 2023-08-06:
Sometime it happened, try reboot and retry.

SaintWe, 2023-08-06:
I have already tried restarting, but it's not working. Do you have any other solutions?

niceboygithub, 2023-08-06:
Skip the update on zb, modify p3_ipdate.sh
'''

COOR_MD5SUM=""
'''

SaintWe, 2023-08-08:
Thank you, I accidentally succeeded after multiple restarts

closed 2023-08-08 by SaintWe