niceboygithub/AqaraM1SM2fw #88: New M2 Gateway Model

miniknife88, 2023-10-22:
Model: ZHWG19LM   Powered by POE

[https://drive.google.com/file/d/11t8XhyjXWaEVwDz6lJTvwAuc93tA6DQR/view?usp=share_link](url)

[https://drive.google.com/file/d/141obdn5baU3V3TFnJP2KqCcV0RfOMZVA/view?usp=share_link](url)

niceboygithub, 2023-10-22:
https://github.com/niceboygithub/AqaraCameraHubfw/blob/main/modified/M2PoE/telnet.md

miniknife88, 2023-10-22:
linux.bin and rootfs.bin  （This Link cannot be opened）

niceboygithub, 2023-10-22:
Not released yet, maybe release after Aqara M3 is on sell

hawkefly, 2024-01-15:
Aqara M3 is on sell now.

niceboygithub, 2024-01-15:
> Aqara M3 is on sell now.

Will you buy one M3 to me?

ashottonoyan, 2024-01-15:
Happy to buy. Where are you based? China, right?

niceboygithub, 2024-01-16:
Not in China.

miniknife88, 2024-01-16:
If possible, would be happy to provide an M3 gateway as a test

ashottonoyan, 2024-01-17:
Ok, let me know where to ship.

closed 2024-01-24 by niceboygithub