niceboygithub/AqaraM1SM2fw #89: [New firmware support]: AIOT_V4.0.6_0006.0646 for `Aqara P3`

Necroneco, 2023-11-13:
```json
  "releaseDate": 1698046109000,
  "part": "main",
  "downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.aircondition.acn05/20230920202318_lumi.aircondition.acn05_AIOT_V4.0.6_0006.0646_20230919_4e0c1a.bin",
  "upgradeTakesTime": 0,
  "updateLog": "1.优化设备相关功能； 2.修复已知问题；",
  "updateTime": 1698046109000,
  "isNecessary": 0,
  "fileSize": 18094184,
  "createTime": 1695892109000,
  "firmwareMD5": "171fd8deff2b2b992d26718b674e0c1a",
  "state": 3,
  "isForce": false,
  "firmwareVersion": "4.0.6_0006.0646",
  "imageType": "bin",
  "firmwareCRC32": "F412A665"
```

Thanks!

Necroneco, 2023-11-14:
Got error...
```
p3_update.sh revision: 2
type: lumi.aircondition.acn05, model: AC_P3
platform: aiot, path: /tmp/fw.tar.gz, sign: 0
Unpack path : /tmp/ota_dir
Firmware path : /data/ota_dir
Available ROM size(b): 18534400
Available RAM size(b): 16285696
Update to 4.0.6_0006.0646
Get packages, please wait...
Got packages done
AalUart_Open /dev/ttyS2
recv STA!
recv STA!
recv close ACK!
recv CRC check!
ir_loacl_crc = 962b, ir_msg_crc=8faf,retry...
recv STA!
Error:Unkonw flag:00000043!
ir_loacl_crc = 0, ir_msg_crc=8faf,retry...
recv STA!
Error:Unkonw flag:00000043!
ir_loacl_crc = 0, ir_msg_crc=8faf,retry...
recv STA!
Error:Unkonw flag:00000043!
ir_loacl_crc = 0, ir_msg_crc=8faf,retry...
recv STA!
Error:Unkonw flag:00000043!
ir_loacl_crc = 0, ir_msg_crc=8faf,retry...
Update failed, reason: OTA failed!
```

niceboygithub, 2023-11-14:
Had you tried again?

Necroneco, 2023-11-14:
Previously, I failed and tried again after rebooting, but same error.    Now I tried again, and it succeeded.  Thanks.