niceboygithub/AqaraM1SM2fw #96: How to use TTL to flash modified firmware to M1S?

bbluvgg, 2024-01-05:
My M1S gateway firmware is 4.0.4_0006, cannot be opened Telnet and use with xiaomi gateway3.

I opened  the case of gateway , Connected  TTL to flash the modified firmware, but aqaragateway.exe displayed a failure：

If flash the latest linux_4.0.8_0006.0650_mi_4.0.4_0006.bin, it will display
[image]

If flash linux_3.2.7_0020.0524.bin, it will display
[image]

Is it my improper operation?I have connected to the GND/RX/TX points near the M1S MCU and provided it with 3.3V power supply.
[image]
[image]

Is there any other way to flash the modified firmware, such as the traditional TTL mode?Is there a detailed tutorial?Thank you!

niceboygithub, 2024-01-06:
You can just flash the 3.x rootfs_0 , donot use 4.x modified firmware to flash cause the size is too big.

niceboygithub, 2024-01-11:
You can use the another method to clear password , please check the gw_global_tool

closed 2024-01-25 by niceboygithub