niceboygithub/AqaraM1SM2fw #103: Flash modified firmware on M1S Gen1 CN

vit5041, 2024-04-04:
Hi! I am thinking to flash M1S Gen1 CN with modified firmware to integrate to HA.. Want to ask, what should I expect to happen with linked devices in Mi Home?

niceboygithub, 2024-04-04:
What do you mean that linked device in Mi Home?

vit5041, 2024-04-04:
I mean already paired sensors in Mi Home, Zigbee child devices.
Some of them are not easily accessible, so I'd like to assess the risk of re-pairing all.

vit5041, 2024-04-06:
So finally I went through it. There is a good chance to keep all paired devices but I did not manage. I had reset GW after making magic clicks for numerous of times to try to open telnet. 
If I would flash modified firmware from the beginning, I would have better chances to keep all linked devices.

closed 2024-07-11 by vit5041