niceboygithub/AqaraM1SM2fw #111: [New firmware support]: AIOT_V4.1.6_0018.0650 for `P3 & M1S`

SaintWe, 2024-07-24:
```
# M1S
https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.acn01/20240624180927_lumi.gateway.acn01_AIOT_V4.1.6_0018.0650_20240624_3a49d8.bin

{
      "deviceModel": "lumi.gateway.acn01",
      "upgradeFirmware": {
        "updateTime": 1720419634000,
        "fileSize": 17705064,
        "state": 3,
        "releaseDate": 1720419634000,
        "imageType": "bin",
        "updateLog": "1.优化设备相关功能；\n2.修复已知问题；",
        "isNecessary": 0,
        "firmwareMD5": "9b23386f00b796dd0527a1bbc63a49d8",
        "isForce": false,
        "part": "main",
        "firmwareCRC32": "3F7319A7",
        "firmwareVersion": "4.1.6_0018.0650",
        "createTime": 1719893731000,
        "downloadUrl": "https:\/\/cdn.aqara.com\/cdn\/opencloud-product\/mainland\/product-firmware\/prd\/lumi.gateway.acn01\/20240624180927_lumi.gateway.acn01_AIOT_V4.1.6_0018.0650_20240624_3a49d8.bin",
        "upgradeTakesTime": 40
      }
}
```

```
# P3
https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.aircondition.acn05/20240625161539_lumi.aircondition.acn05_AIOT_V4.1.6_0018.0650_20240625_44f63b.bin

{
      "deviceModel": "lumi.aircondition.acn05",
      "upgradeFirmware": {
        "updateTime": 1720426332000,
        "fileSize": 18554984,
        "state": 3,
        "releaseDate": 1720426332000,
        "imageType": "bin",
        "updateLog": "1.优化设备相关功能； 2.修复已知问题；",
        "isNecessary": 0,
        "firmwareMD5": "e4ef1c7fdc3bb08254c72a579b44f63b",
        "isForce": false,
        "part": "main",
        "firmwareCRC32": "024C12AC",
        "firmwareVersion": "4.1.6_0018.0650",
        "createTime": 1719893928000,
        "downloadUrl": "https:\/\/cdn.aqara.com\/cdn\/opencloud-product\/mainland\/product-firmware\/prd\/lumi.aircondition.acn05\/20240625161539_lumi.aircondition.acn05_AIOT_V4.1.6_0018.0650_20240625_44f63b.bin",
        "upgradeTakesTime": 0
      }
}
```

closed 2024-07-30 by SaintWe

vit5041, 2024-07-31:
Dear SaintWe, could you please also check if there is FW for Aqara E1 hub, for the https://github.com/niceboygithub/AqaraCameraHubfw

SaintWe, 2024-07-31:
@vit5041 Sorry, I need to have the relevant device in order to obtain the firmware