niceboygithub/AqaraM1SM2fw #116: ZNDMWG04LM does not reset password

dmitkam, 2024-09-22:
ZNDMWG04LM does not reset password firmware 1.0.7_0019

== Gateway Global Tool version:20240218==

   >>Power ON Gateway NOW!<<

IPL gdf99011
D-17
HW Reset
01481480 00000000
Resume? N, addr 00000000
miupll_233MHz
SPI 54M
128MB
BIST0_0001-OK
SPI 54M
[BBT] Found table @ 0x00020000

Checksum OK

IPL_CUST gdf99011
Export ENV 1

U-Boot 2015.01 (Sep 22 2022 - 15:39:48)

Version: P3g1fd806f
I2C:   ready
DRAM:
WARNING: Caches not enabled
SPINAND_I:  [FLASH] Found SNI in block 0.
[FLASH] dev_id = 0xee
[FLASH] mfr_id = 0xa1, dev_id= 0xe4 id_len = 0x2
[SPINAND] RFC ues command 0x6b with 0x08 dummy clock.
[SPINAND] Program load with command 0x32.
[SPINAND] Random load with command 0x34.
[FLASH] Unlock all block.
[FLASH] Use BDMA.
128 MiB
MMC:   MStar SD/MMC: 0
ENV: offset = 0x480000 size = 0x40000
ENV1: offset = 0x4c0000 size = 0x40000
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
clk=12M, u16Div=0 u32Duty=0x2cf u32Period=0x4af
[halPWMPadSet][107] (pwmId, padId) = (1, 5)
clk=12M, u16Div=0 u32Duty=0x4af u32Period=0x4af
[halPWMPadSet][107] (pwmId, padId) = (2, 6)
clk=12M, u16Div=0 u32Duty=0x4af u32Period=0x4af
[halPWMPadSet][107] (pwmId, padId) = (3, 7)
gpio debug MHal_GPIO_Pad_Set: pin=43
gpio[43] is 1
gpio debug MHal_GPIO_Pad_Set: pin=44
gpio[44] is 1
gpio debug MHal_GPIO_Pad_Set: pin=59
gpio[59] is 0
gpio debug MHal_GPIO_Pad_Set: pin=62
gpio[62] is 1
gpio debug MHal_GPIO_Pad_Set: pin=63
gpio[63] is 0
gpio debug MHal_GPIO_Pad_Set: pin=61
gpio[61] is 1
gpio debug MHal_GPIO_Pad_Set: pin=60
gpio[60] is 1
gpio debug MHal_GPIO_Pad_Set: pin=44
gpio[44] is 0
gpio debug MHal_GPIO_Pad_Set: pin=63
gpio[63] is 1
gpio debug MHal_GPIO_Pad_Set: pin=59
[... 310 lines cut ...]
module [ai] init
ubiattach /dev/ubi_ctrl -m 10  -d 0
UBI device number 0, total 160 LEBs (20316160 bytes, 19.4 MiB), available 0 LEBs
ubiattach /dev/ubi_ctrl -m 11  -d 1
chan change ch 6, type 1, off_chan 0
INIT SSTAR CONTROL GENERIC NETLINK MODULE
UBI device number 1, total 479 LEBs (60821504 bytes, 58.0 MiB), available 0 LEBs
[WatchDogInit 15] init watch dog, timeout:30s
skip
/ # passwd -d root
passwd: unknown uid 0
/ #
/ # cat /data/miio/device.token
WLv59RA0Lwk5x7UH
/ # Gateway Info:

cat /data/miio/device.conf
did=499480362
key=kWHyh7Yh1IovFjhn
mac=54:EF:44:48:60:A9
vendor=lumi
model=lumi.gateway.mgl001
/ #
/ # reboot
/ #

niceboygithub, 2024-09-22:
You got key, that is enough for XG3

dmitkam, 2024-09-22:
Telnet is closed. Custom component for control Xiaomi Multimode Gateway (aka Gateway 3), Xiaomi Multimode Gateway 2, Aqara Hub E1 - https://github.com/AlexxIT/XiaomiGateway3 - can't connect. You can't telnet via PuTTY.

niceboygithub, 2024-09-22:
The telnet will be enabled by XG3 via token and key.

dmitkam, 2024-09-28:
Apparently the root user is missing. )

niceboygithub, 2024-09-28:
mgl001 use root as user.

dmitkam, 2024-09-28:
If the key and Token are present, telnet will not open. The creator of the XiaomiGateway3 integration assumes that there is no root user.

niceboygithub, 2024-09-29:
The "root" user was NOT removed in every firmware till now.