niceboygithub/AqaraM1SM2fw #121: Is it possible to let us generate modified firmware ourselves?

tidegu, 2024-10-15:
Many thanks for your work here! @niceboygithub 
Ver 4.1.8 has been released for a while, and from discussions in issue #115, I'm not sure if you'll release a modified firmware anytime soon. I wonder how difficult if we generate the modified firmware ourselves, can you possibly give some sources and instructions so that we can pack our own firmwares?
Not sure if the modification is case by case or there's some scripts to do it automatically, if latter is the case, it would be very much appreciated if you can share the instructions so that we can catch up the latest firmware and stop worrying if we accidentally updated our devices - I have ~10 gateways running in my house, it would be nightmare if an accident upgrade is made, especially I will have to flash every single device using TTL then...

niceboygithub, 2024-10-16:
1. get original rootfs.
2. unsquashfs the rootfs with root
3. replace mosquitto which is using public 1883 port
4. enable telnetd in /etc/init.d/rcS
5. pack the rootfs with command 'mksquashfs ./squashfs-root rootfs.squashfs -comp xz -always-use-fragments -all-root -no-xattrs -noappend'

tidegu, 2024-10-17:
BTW, here are some urls for the latest update, if it's convenient, is that possible for you to pack the new versions fw? @niceboygithub 

M1S(2021):
```
"deviceModel": "lumi.gateway.acn01",
"upgradeFirmware": {
	"releaseDate": 1726652369000,
	"part": "main",
	"downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.acn01/20240827164453_lumi.gateway.acn01_AIOT_V4.1.8_0016.0650_20240827_0b6920.bin",
	"upgradeTakesTime": 40,
	"updateLog": "1、修复偶现子设备固件在App上反复提示升级的问题\n2、修复自动化的执行动作配置为其他自动化开/关状态时，自动化为非本地自动化的问题 \n3、修复其他已知问题",
	"updateTime": 1726652369000,
	"isNecessary": 0,
	"fileSize": 17756264,
	"createTime": 1725855781000,
	"firmwareMD5": "77bc6d9f7bb2a4b524747d4c5e0b6920",
	"state": 3,
	"isForce": false,
	"firmwareVersion": "4.1.8_0016.0650",
	"imageType": "bin",
	"firmwareCRC32": "A647B3C1"
},
```

P3:
```
"deviceModel": "lumi.aircondition.acn05",
"upgradeFirmware": {
	"releaseDate": 1726729249000,
	"part": "main",
	"downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.aircondition.acn05/20240902183643_lumi.aircondition.acn05_AIOT_V4.1.8_0018.0650_20240830_6f676f.bin",
	"upgradeTakesTime": 0,
	"updateLog": "1、修复偶现子设备固件在App上反复提示升级的问题 \n2、修复自动化的执行动作配置为其他自动化开/关状态时，自动化为非本地自动化的问题 \n3、修复其他已知问题",
	"updateTime": 1726729249000,
	"isNecessary": 0,
	"fileSize": 18657384,
	"createTime": 1726129613000,
	"firmwareMD5": "12483f2b59b641b12ffd180e3d6f676f",
	"state": 3,
	"isForce": false,
	"firmwareVersion": "4.1.8_0018.0650",
	"imageType": "bin",
	"firmwareCRC32": "91A2C678"
},
```

H1:
```
"deviceModel": "lumi.gateway.sacn01",
"upgradeFirmware": {
	"releaseDate": 1727073457000,
	"part": "main",
	"downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.sacn01/20240827160339_lumi.gateway.sacn01_AIOT_V4.1.8_0016.0750_20240827_73c848.bin",
	"upgradeTakesTime": 40,
	"updateLog": "1、修复偶现子设备固件在App上反复提示升级的问题\n2、修复自动化的执行动作配置为其他自动化开/关状态时，自动化为非本地自动化的问题 \n3、修复其他已知问题",
	"updateTime": 1727073457000,
	"isNecessary": 0,
	"fileSize": 16599144,
	"createTime": 1725864104000,
	"firmwareMD5": "1bcb77d9d852f2949ad7cd341f73c848",
	"state": 3,
	"isForce": false,
	"firmwareVersion": "4.1.8_0016.0750",
	"imageType": "bin",
	"firmwareCRC32": "76BCD2D9"
},
```

M1S (2022) - I use it just for testing new devices, so I don't need to patch it anyway. Just for in case other users may need a modified fw for this model:
```
"deviceModel": "lumi.gateway.acn004",
"upgradeFirmware": {
	"releaseDate": 1726652139000,
	"part": "main",
	"downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.acn004/20240909102456_lumi.gateway.acn004_AIOT_V4.1.8_0016.0022_20240906_0e2b47.bin",
	"upgradeTakesTime": 40,
	"updateLog": "1、修复偶现子设备固件在App上反复提示升级的问题\n2、修复自动化的执行动作配置为其他自动化开/关状态时，自动化为非本地自动化的问题\n3、修复其他已知问题",
	"updateTime": 1726652139000,
	"isNecessary": 0,
	"fileSize": 15339624,
	"createTime": 1726129541000,
	"firmwareMD5": "25aa25d04147b10e97be1d74540e2b47",
	"state": 3,
	"isForce": false,
	"firmwareVersion": "4.1.8_0016.0022",
	"imageType": "bin",
	"firmwareCRC32": "A1EEAD35"
},
```

closed 2024-10-19 by tidegu

cyanogenic, 2025-06-06:
@tidegu How did you get those download URLs?
Looking forward to your guidance!