niceboygithub/AqaraM1SM2fw #125: Firmware minor version mismatch

tidegu, 2024-12-15:
Hi @niceboygithub , many thanks for your work, but just wondering why sometimes after upgrading using telnet, the minor version of the firmware shows differently from Aqara Home app? 

This happened before on my M1s, but after the next upgrade (or by flash using UART I don't remember), the issue resolved, so I didn't care about it.

Today I got a new H1 and flashed it using UART to version 4.1.4_0008.0750 and then updated it using telent, it shows version 4.3.0_0008.0746, instead it supposed 4.3.0_0008.0750. Tried to update again using telnet, the green text shows 0750 but after reboot, it still shows 0746 from AH app.

I don't know the FW version before flashing - I bought it second hand and didn't even bother to connect to the App, just straightly crack the case open and flash.

I didn't flash it to the latest FW for two reasons:
 - one, lazy to update my local file - I had the whole repo downloaded as a zip when the first time learning hacking Aqara device to HA, the latest modified FW by then was 4.1.4, I didn't bother to update my local file, because I anyway will update using telnet again to make sure it works
 - two, I always try to make all my devices consistant. I don't know if the outcome will be same for flashing or telnet-updating, but I guess flashing is more deep and clean but I'd rather prefer keep all my devices on the same track rather than having slight version difference, in case any bug happens it would be a nightmare to diagnose.

Thanks in advance for your kindest help.

niceboygithub, 2025-02-14:
That means that the zigbee firmware upgraded failed.
Try to power off the gateway for several minutes and power on again. Then try to update again.

tidegu, 2025-03-13:
> That means that the zigbee firmware upgraded failed. Try to power off the gateway for several minutes and power on again. Then try to update again.

Good point! Thanks! Since it's H1, it's hard to get power disconnected. Rebooting didn't help IIRC. I'll keep my eye on it if this issue happens again in other devices and troubleshot along with your solution. Thanks!

closed 2025-03-13 by tidegu