niceboygithub/AqaraM1SM2fw #128: [New firmware support] AIOT_V4.3.2_0026.0652 for P1 and H1 (maybe)

tidegu, 2025-02-14:
```
"deviceModel": "lumi.aircondition.acn05",
"upgradeFirmware": {
    "releaseDate": 1739447305000,
    "part": "main",
    "downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.aircondition.acn05/20250210180811_lumi.aircondition.acn05_AIOT_V4.3.2_0026.0652_20250210_cffe37.bin",
    "upgradeTakesTime": 0,
    "updateLog": "Fixed known issues",
    "updateTime": 1739447305000,
    "isNecessary": 0,
    "fileSize": 18585704,
    "createTime": 1739330410000,
    "firmwareMD5": "d610f2b77c8ae1045530cece93cffe37",
    "state": 3,
    "isForce": false,
    "firmwareVersion": "4.3.2_0026.0652",
    "imageType": "bin",
    "firmwareCRC32": "0F0F240F"
},
```

I also have several H1 at home, but only one of them giving me update (V4.3.2_0026.0752) as below - this one is on 4.3.0_0008.0746 while others are  on4.3.0_0008.0750 (see #125).

```
"deviceModel": "lumi.gateway.sacn01",
"upgradeFirmware": {
    "releaseDate": 1739334595000,
    "part": "main",
    "downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.gateway.sacn01/20250210172417_lumi.gateway.sacn01_AIOT_V4.3.2_0026.0752_20250210_b4b2e7.bin",
    "upgradeTakesTime": 40,
    "updateLog": "1. Optimize device related functions; 2. Fix known issues;",
    "updateTime": 1739334595000,
    "isNecessary": 0,
    "fileSize": 16527464,
    "createTime": 1739334595000,
    "firmwareMD5": "4117346ffa3b4faf4b40e1d4c2b4b2e7",
    "state": 2,
    "isForce": false,
    "firmwareVersion": "4.3.2_0026.0752",
    "imageType": "bin",
    "firmwareCRC32": "0BA85AB6"
},
```

closed 2025-02-17 by tidegu

ritech, 2025-02-22:
How did you get this information? Telnet in and execute a command?

ritech, 2025-02-22:
Can you provide the latest m1s 2021 firmware file download link?