niceboygithub/AqaraM1SM2fw #136: Request: How to patch the firmware – I'd like to contribute

cyanogenic, 2025-06-06:
Hi,

Thank you for your great work on this project! It's really helpful!

Would you be willing to share the method or tools you used to apply the patch? I'd love to learn how it's done so I can also try patching the firmware myself and potentially contribute to the project.

Looking forward to your guidance. Thanks again!

Best regards

niceboygithub, 2025-06-06:
1. Unsquashfs rootfs
2. add telnetd in /etc/init.d/rcS
3. replace the mosquitto which support public port 1883
4. mksquashfs

cyanogenic, 2025-06-06:
> 1. Unsquashfs rootfs
> 2. add telnetd in /etc/init.d/rcS
> 3. replace the mosquitto which support public port 1883
> 4. mksquashfs

Is binutils/mosquitto OK?

niceboygithub, 2025-07-04:
No, it contains the header, size, etc